rust-embedded/rust-raspberrypi-OS-tutorials #11: Mailbox access not working after MMU init

andre-richter, 2019-01-21:
It looks like I/O with the mailbox is not working after `mmu::init`. This needs to be investigated.

andre-richter, 2019-01-22:
Ah, that sounds very likely. I'll ponder a bit what a good solution in the scope of the tutorials could look like.

Thanks!

jakelstr, 2019-01-23:
I just wanted to add that I was experimenting with the framebuffer on top of the rust code, and I ran into the exact same issue where it wouldn't work unless I put in a lot of writes to the UART. I assumed it was my fault.

andre-richter, 2019-01-23:
A quite easy workaround for now for you guys would be to disable data caching altogether in `mmu.rs`: `SCTLR_EL1::C::NonCacheable`.

I have yet to decide how I want to introduce this topic in a way that makes sense both didactically and is not completely out of touch of how this stuff is approached in the real world. So far I thought about:
- Declaring such buffers global static and link them into a non-cacheable page.
- Introduce a special allocator that allocates from a non-cacheable page at runtime.
- Make use of flush and invalidate operations before/after exchange with the Videocore.

andre-richter, 2019-01-27:
FYI, I have decided to go the allocator way, since this one comes closest to real world kernels where you would use functions called like `dma_alloc()` to obtain a piece of memory that can be used for these purposes. Hopefully I'll get it done within the next few days.

I could also reproduce atomics not working when data chache is disabled. I was actually surprised to find out about that.

digitalstreamio, 2019-01-27:
Sounds good. I look forward to seeing how that would work with the allocator setup. Would you need two separate allocators, one for heap and one for device memory?

andre-richter, 2019-01-28:
Here is a sneak peak: https://github.com/rust-embedded/rust-raspi3-tutorial/tree/println_dma/0F_DMA_memory

This is all still subject to change and cleanups which will happen with force pushes at any time. Also need to yet write the prose.

Output currently only works on a real RPi, because I haven't come up with a solution yet to multiplex two virtual UARTs to STDIO using QEMU.

Idea is to rebase the exception lesson on top of the DMA_memory lesson, then add a lesson that transforms the NullLocks to real Spinlocks, then enable Interrupts:
- 0E println
- 0F DMA memory
- 10 Exceptions
- 11 Spinlocks and Critical Sections
- 12 Interrupts

Cheers,
Andre

jakelstr, 2019-01-28:
If one of you has a moment, could you explain why atomics don't work without data cache, or possibly direct me to something to read? I'm fairly new to this advanced level and I'm trying to learn all I can.

andre-richter, 2019-01-28:
Unfortunately, this is a very complicated topic in terms of processor architecture, very dependent on the implementer of the SoC and on top of that, poorly documented.

On AArch64, exclusive access operations are realized using so called monitors.
There exist two of them, `local monitors` and `global monitors`.  I recommend you glimpse into the ARMv8 Arch Reference Manual, and additionally read https://stackoverflow.com/questions/23914222/arm-is-ldrx-strx-needed-if-interrupts-are-disabled.

What I can say is, if you execute exclusive store/load instructions on the RPi3 with exceptions enabled but MMU disabled, you get a synchronous abort (you can test it by modifying lesson `0E`):

```console
A synchronous exception happened.
    ELR_EL1: 0x80584
    ESR_EL1: 0x96000035

$ make objdump | rg 80584
   80584:       6e 7d 5f c8     ldxr    x14, [x11]
```

Upon further inspecting the given `ESR_EL1`, you can see that `ESR_EL1.DFSC = 110101`, which is `IMPLEMENTATION DEFINED fault (Unsupported Exclusive or Atomic access)`.

Please inspect Pages 132 and 2463 of the ARMv8 Arch reference manual where these decodings are reported.

andre-richter, 2019-02-01:
Fixed by https://github.com/rust-embedded/rust-raspi3-tutorial/commit/48cf71b6d49d85629acf8d7aaa8385e41557b9dd.

@digitalstreamio @jakelstr FYI.

closed 2019-02-01 by andre-richter

digitalstreamio, 2019-02-02:
I really like your separate dma allocator setup. The code and its structure look great. It keeps with the clean and easy to follow spirit of other lessons.

I have one question on the memory map that you may be able to shed some light on. I'm a bit confused on when to put stuff into linker definition vs hard code in the source code. Any thoughts on that?

Also, I'm looking forward to your synchronization tutorial.  I would love to see it bring up multicore setup.

Thanks