rust-embedded/rust-raspberrypi-OS-tutorials #12: Add `+strict-align` to `.cargo/config`

andre-richter, 2019-01-21:


andre-richter, 2019-02-02:
@digitalstreamio this is now upstream and should land in the nightly toolchain sometime soon.

Prevents #10

closed 2019-02-02 by andre-richter

digitalstreamio, 2019-02-02:
Awesome. I was just hit by this again while trying to copy data from device memory used by my sd card handling code to my app level state. 

I can certainly appreciate the generated code no longer halting the cpu ;).

Out of curiosity though, are there any downsides to turning this on?

digitalstreamio, 2019-02-02:
Also, I will gladly give it a spin in a day or two once nightly picks it up. Cheers!

andre-richter, 2019-02-03:
With the option turned on, the compiler circumvents miss-aligned accesses by generating code in a way that it doesn't fail, eg using two 32bit accesses instead of one 64bit access if your operate on an address where modulo 8 != 0. This will cause your code to function, but probably it won't be the fastest possible code.

So you gain  that your code doesn't crash and  trade It for potentially slight performance hits.