rust-embedded/rust-raspberrypi-OS-tutorials #13: _boot_cores.rs and multiple cores

artit91, 2019-02-09:
Hi,

I'm quite new in embedded development and I had to figure out how to start all of the cores.

```
 const CORE_1_START: *mut u32 = (0xe0) as *mut u32;
 const CORE_2_START: *mut u32 = (0xe8) as *mut u32;
 const CORE_3_START: *mut u32 = (0xf0) as *mut u32;

 write_volatile(CORE_1_START, 0x80000);
 write_volatile(CORE_2_START, 0x80000);
 write_volatile(CORE_3_START, 0x80000);
```

Hope it helps someone, it would be nice if I could read somewhere.
(I spent hours because I read an outdated documentation)

Cheers,
Attila

jamesmunns, 2019-02-10:
Hey @artit91, could you please submit this information as a PR where you would have liked to see that? We'd love to add this information to the tutorial!

andre-richter, 2019-02-10:
Hi Attila, James,

It is indeed planned to have a tutorial on using multiple cores concurrently, but it will take some time until we get there. We need to write a scheduler first, and have yet to cover some tutorials that are dependencies for having a scheduler (interrupts, spinlocks and potentially other synchronization primitives).

Also, the code you posted puzzles me a bit. Can you post a reference?

What we do right now is parking all cores except core0 with the `wfe` (wait for event) machine instruction, and only let core0 process further with an initialization routine that is tailored for it.
As a sneak peek: Once we get there with our tutorials, we can wake up the remaining slave cores by sending them an "event" and let them come online with their own init routine once core0 has set up the scheduler and the likes.
You can read more about `wfe` and sending events in the ARMv8 architecture reference manual.

BR,
Andre

artit91, 2019-02-11:
Hi,

I don't want to leave you confused, I have tried the code in qemu and it didn't start all the cores.

Then I have found this:
https://github.com/qemu/qemu/blob/595123df1d54ed8fbab9e1a73d5a58c5bb71058f/hw/arm/raspi.c#L67

From the code I have figured out that I have to write to those addresses to able to start the cores.

I haven't tested on a real hardware though.

BR,
Attila

andre-richter, 2019-02-11:
Thank you. With the new context it makes sense. What you see is the same as I laid out earlier.

The secondary cores spin on `wfe` until an event occurs, and then they check predefined memory locations (`0xe0`, ...) for magic values that tell them to either go back to sleep again or jump to boot code for secondary cores.

What you are pointing at is other firmware, and not some HW functionality, that's why it doesn't work for you now, because we are putting our secondary cores into infinite sleep as of now: https://github.com/rust-embedded/rust-raspi3-OS-tutorials/blob/master/07_abstraction/raspi3_boot/src/lib.rs#L85-L88

 When we will have our multi-core tutorial sometime in the future, this scheme will be used as well.

andre-richter, 2019-02-13:
Hi Attila,

is this enough information for you to close this issue, or do you have any follow up questions?

BR,
Andre

artit91, 2019-02-13:
Hi,

Yes, enough.

Thank you,
Attila

closed 2019-02-13 by artit91