rust-embedded/rust-raspberrypi-OS-tutorials #18: Improve JTAG tutorial

andre-richter, 2019-03-09:
- [ ] To the JTAG fritzing wiring image, add a red dot indicating the LED on the Olimex HW for easier orientation
- [ ] OpenOCD spews lots of errors while debugging. Check if they are harmless or can be fixed.
- [ ] CPSR in GDB seems to show wrong `EL`. Maybe other fields as well?
- [ ] Build GDB from source in the Docker container to be more bleeding edge regarding Rust support?
- [ ] The real-life wiring image has been flipped to horizontal from originally being vertical, in order to save space. It looks distored now. Take a new "native" horizontal version?
- [ ] Maybe more?

CC @naotaco if you like to help.

berkus, 2019-03-10:
I'm actually using `rust-gdb` wrapper from rust and specify the GDB binary to the one i built/installed myself. Rust-gdb loads some rust-related macros for easier debugging.

Do you do the same?

naotaco, 2019-03-10:
I hope these images are slightly better than current one.  Please feel free to take them if you like, or should I open new PR to replace?

[image]
[image]

naotaco, 2019-03-10:
@berkus

No I don't. I just use `aarch64-elf-gdb` from [Arm](https://developer.arm.com/open-source/gnu-toolchain/gnu-a/downloads). 

Does it work with Rpi3? (maybe this one? https://github.com/rust-lang/rust/blob/master/src/etc/rust-gdb)

andre-richter, 2019-03-10:
> I'm actually using rust-gdb wrapper from rust and specify the GDB binary to the one i built/installed myself. Rust-gdb loads some rust-related macros for easier debugging.

Thats maybe a worthy addition to the current gdb container. I wasn't sure if it wraps around self-built gdbs.

What are the benefits besides from pretty printing? Does the pretty-printing work in conjunction with the dashboard?

andre-richter, 2019-03-10:
> 
> 
> I hope these images are slightly better than current one. Please feel free to take them if you like, or should I open new PR to replace?

PR would be nice :)

berkus, 2019-03-11:
@andre-richter I haven't used gdb-dashboard much, just turned it on after seeing your JTAG tutorial :) Seems to still work fine, but I'll keep an eye on it for a while.

rust-gdb accepts RUST_GDB environment var as an alternative gdb path [like here](https://github.com/metta-systems/vesper/blob/mmu-enable/Makefile#L96).

@naotaco yeah, you can wrap that one using RUST_GDB envvar and let rust-gdb load appropriate rust macros for you, good part is that rust-gdb is part of rust installation and rust team cares about what current rust stuff must be included, so you don't need to think about it much.

naotaco, 2019-03-11:
I tried it and found rust-gdb runs smoothly. But it was too natural for me; honestly, I couldn't find difference between rust-gdb and `gdb-multiarch` that used in the current gdb container even without the dashboard... 

The command I used is like `RUST_GDB=gdb-multiarch rust-gdb -q kernel8_debug`

 It would be appreciated if you could explain a situation that good to find the advantage. Thanks in advance.

andre-richter, 2019-03-11:
Just tested as well, also couldn't make out a striking difference yet.

berkus, 2019-03-12:
Not a problem, there's no striking difference at the moment. But in case there is more gdb support bundled with rust I prefer to future-proof as let rust team manage this, as I said above. In case any significant magic is added in the future mine will just work with `rustup update`.

andre-richter, 2019-03-13:
Yah, in the long run it makes sense to use it.
In case of the tutorials, I think we need to bind-mount the local Rust installation into the container so that we ensure to have the same version that we compiled, put it into container's path and execute rust-gdb.

andre-richter, 2019-03-13:
@berkus Does the JLink stuff work as well or are you yet to test it?

berkus, 2019-03-16:
Have not tested yet but will do it right now! Been a busy week.

berkus, 2019-03-16:
The default `make openocd` command fails with `Error: unable to open ftdi device with vid 15ba, pid 002a, description 'Olimex OpenOCD JTAG ARM-USB-TINY-H', serial '*' at bus location '*'`

So it doesn't "just work", need to see how the docker image built to override this default.

berkus, 2019-03-16:
Yeah, afaiu this means rebuilding the docker image with different setting `-f /openocd/tcl/interface/jlink.cfg`, to avoid this the best approach would be to [use a combination of ENTRYPOINT and CMD](https://medium.freecodecamp.org/docker-entrypoint-cmd-dockerfile-best-practices-abc591c30e21) options when creating the docker image, allowing one to override the interface.

I.e.
```
ENTRYPOINT ["openocd", "-f", "/openocd/rpi3.cfg"]
CMD ["-f", "/openocd/tcl/interface/ftdi/olimex-arm-usb-tiny-h.cfg"]
```

Which would allow to simply override the interface by doing `docker run ... -f /openocd/tcl/interface/jlink.cfg`

berkus, 2019-03-16:
This is implemented in [this PR](https://github.com/rust-embedded/rust-raspi3-OS-tutorials/pull/20).

andre-richter, 2019-03-16:
When I tried the new container locally, it broke the current setup because it seems that it matters in which order the `-f` arguments are passed to openocd.

`-f /openocd/rpi3.cfg -f  /openocd/tcl/interface/ftdi/olimex-arm-usb-tiny-h.cfg` <-- fails
`-f  /openocd/tcl/interface/ftdi/olimex-arm-usb-tiny-h.cfg -f /openocd/rpi3.cfg ` <-- works

berkus, 2019-03-17:
Hmm, that's bad. We could of course make the entire command line overridable, will be more messy for non-standard JTAGs, but I guess that's life.

berkus, 2019-03-17:
^ Lets try #22 instead then.

berkus, 2019-03-17:
Ok, now I can try the tutorial with new openocd docker image.

berkus, 2019-03-17:
With docker image on osx it for some reason could not find the USB device.
```
λ make openocd
docker run -it --rm --privileged -v /dev:/dev --network host andrerichter/raspi3-openocd -f /openocd/tcl/interface/jlink.cfg -f /openocd/rpi3.cfg
Open On-Chip Debugger 0.10.0+dev-g8f518d3 (2019-03-16-15:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
trst_and_srst separate srst_gates_jtag trst_push_pull srst_open_drain connect_deassert_srst
adapter speed: 1000 kHz
jtag_ntrst_delay: 500
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: No J-Link device found.
```
This needs more investigation.

andre-richter, 2019-03-18:
@berkus 
It seems this is quite fiddly in OSX nowadays =/

https://milad.ai/docker/2018/05/06/access-usb-devices-in-container-in-mac.html

berkus, 2019-03-18:
Oh well, I guess I can skip the container part, since native stuff works well on osx.

rahealy, 2019-10-11:
I can confirm that:

```
$ docker run -it --rm --privileged -v /dev:/dev --network host andrerichter/raspi3-openocd -f openocd/tcl/interface/ftdi/olimex-jtag-tiny.cfg -f /openocd/rpi3.cfg
```

Connects to my old Olimex ARM-USB-TINY (No -H) but fails with the following output:

```
Open On-Chip Debugger 0.10.0+dev-g8f518d3 (2019-03-16-15:37)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
trst_and_srst separate srst_gates_jtag trst_push_pull srst_open_drain connect_deassert_srst
adapter speed: 1000 kHz
jtag_ntrst_delay: 500
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: rpi3.tap: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Invalid ACK (7) in DAP response
Error: JTAG-DP STICKY ERROR
...
```
I've triple checked my wiring. I suspect the JTAG adapter needs to be retired and replaced. At this point I'm unable to contribute more until I can get a working JTAG adapter.

andre-richter, 2019-10-11:
@rahealy just to confirm, have you followed the steps in https://github.com/rust-embedded/rust-raspi3-OS-tutorials/tree/master/0B_hw_debug_JTAG#in-summary exactly?

To be honest I haven't tried it in a long time, but I remember that even the order of connecting cables was crucial.

rahealy, 2019-10-12:
> just to confirm, have you followed the steps in https://github.com/rust-embedded/rust-raspi3-OS-tutorials/tree/master/0B_hw_debug_JTAG#in-summary exactly?

No I have not. I'm only confirming specifically the ability to pass command line options to use different OpenOCD compatible JTAG adapters as implemented by @berkus  in #22.

I've fought with this particular JTAG adapter and OpenOCD for years. The fact that it doesn't work on the first try doesn't surprise me. I will follow the steps exactly as written in the tutorial and report back.

rahealy, 2019-10-12:
Followed instructions with exception of the changes outlined in #31. Same results. I'm going to order a new JTAG adapter but likely won't get to it this month. Thank you for your work :)

andre-richter, 2019-10-28:
Note: As of yesterday, I merged the rewrite of the tutorials into the master branch, and also pushed the updated version of the JTAG tutorial.

What's interesting is, that on the Raspberry Pi 4, the approach of setting the JTAG pins from SW from the CPU was not working out for me. I need to enable JTAG through the `config.txt`, otherwise it would not work on the RPi 4. To double check, I dumped all the GPIO registers after booting with the `config.txt` option enabled and then replicated the same state from the kernel, but OpenOCD would not connect this way.

It seems that on the Raspberry Pi 4, the Videocore is turning some additional invisible knobs in the background.

Relying on users to use the `config.txt` option now.

berkus, 2019-11-03:
I guess that's fine, I have `config.txt` method written down as recommended for debugging anyway.

andre-richter, 2020-04-16:
Closing for now since open points are unlikely to be addressed anytime soon.

closed 2020-04-16 by andre-richter