rust-embedded/rust-raspberrypi-OS-tutorials #36: Uart don't work on RPi4

corco, 2019-11-13:
I am not able to make the UART example (06) work on my Raspberry Pi 4.

I started from an up-to-date Raspbian SD Card, deleted everything from the boot partition except:

- start4.elf
- fixup4.dat
- bcm2711-rpi-4-b.dtb (isn't this a Linux file? why is it needed?)

Then compiled example 06 with `BSP=rpi4 make` and copied kernel8.img to the SD. I also created the config.txt with content:

> init_uart_clock=48000000

I know the UART cable is working since I enabled the bootloader UART (as per https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) and I'm getting the bootloader output.

However, I do not get any output from the Rust code, and I'm not sure how to debug. The last lines of the bootloader are:

> Loading start4.elf hnd: 0x0002e101
> Loading fixup4.dat hnd: 0x00033c3d
> MEM GPU: 76 ARM: 948 TOTAL: 1024
> FIXUP src: 128 256 dst: 948 1024
> Starting start4.elf @ 0xfec00200

corco, 2019-11-13:
I found my own issue. My terminal was configured to send '\r' on enter instead of the required '\n'.

I did confirm that `bcm2711-rpi-4-b.dtb` is not needed on my board. It also works without `config.txt`.

closed 2019-11-13 by corco

andre-richter, 2019-11-13:
> I did confirm that bcm2711-rpi-4-b.dtb is not needed on my board.

Interesting, I did several experiments with the clear outcome that the dtb blob is needed. I was not starting from a raspian image, though, but from the official firmware folder as described in the tut.

Anyways, its the safer call to have the file, since a stale file doesn't hurt, but interesting still that it works for you without it.