rust-embedded/rust-raspberrypi-OS-tutorials #38: The tutorials begs for this question : Could the information be used to make a Redox kernel?

larsth, 2019-12-01:
I know that the [Redox OS](https://www.redox-os.org/)  [only supports the x86-64 architecture](https://doc.redox-os.org/kernel/kernel/#the-redox-os-kernel-version-2), but using a Redox kernel on a Raspberry Pi Zero (w/wo WiFi) could be really interesting.

I know that a Raspberry Pi Zero has a ARM1176JZF-S (32-bit) ARMv6 CPU, but it has 512 MB of memory, and that is a LOT of memory in the embedded world, and when there is the price tag (US$ 10 in the US) for a RPI Zero ...

larsth, 2019-12-01:
Side note:
Ultibo(1) contains interesting Pascal (FreePascal) source code for creaitng an embedded system using a Raspberry Pi

Ad (1)
[Ultibo](https://ultibo.org/)

andre-richter, 2019-12-01:
These are very different things going here.

Redox is a full fledged OS developed to be actually used on consumer hardware.

The tutorials here are developing a very minimalistic kernel with focus on educational aspects wrt to certain OS building blocks rather than end-user usability. They should teach and serve as a minimalistic reference.

People in the past started a [proof of concept for aarch64](https://archive.fosdem.org/2019/schedule/event/microkernel_written_in_rust/) Redox.

However, both this and the tutorials here are for 64bit ARM. 32 Bit ARM architecture is very different, so no 1:1 reuse possible.

larsth, 2019-12-01:
Thanks for the explanation.

closed 2019-12-01 by larsth