rust-embedded/rust-raspberrypi-OS-tutorials #39: Chinese version

colachg, 2020-01-13:
Hi, @andre-richter do you mind if I translate your project to Chinese? I think this is worth spreading.

andre-richter, 2020-01-13:
Hi @colachg,

It should be possible to arrange that. The question is where and how to do it.
Are you interested in `README.md` translation only, or also comments in source code?

@rust-embedded/resources I think it would make sense to keep a translation under the WG, what do you think?

Approaches I can think of is:

- Add a `README.md.cn` version to each tutorial (works only if only `README.md` is to be translated).
- Have a `chinese` branch to runs in parallel to `master`. This way source code comments could also be translated.

colachg, 2020-01-13:
I think `README.md ` is enough and source code comment is not necessary. Sometimes translator works better. People should know what we are doing, how to do it and why it should be this.
Maybe there should be another voice :)

therealprof, 2020-01-13:
@colachg That's a great idea. I think the only obstacles here are technical ones, like:
* How to present it
* How to verify the correctness (only very few of us read Mandarin, probably close to none 😅)
* How to maintain it, i.e. make sure that updates to the English version are reflected in the Chinese version

andre-richter, 2020-01-13:
> I think README.md is enough and source code comment is not necessary. 

Great. In that case, adding a `README.CN.md` to the repository on `master` will probably ensure the best visibility.
We can also add to the repo's main README.md a hint about their existence.

> How to maintain it, i.e. make sure that updates to the English version are reflected in the Chinese version

This is rather difficult I guess. I can try to ping people when I do updates to the english version.

colachg, 2020-01-13:
> This is rather difficult I guess. I can try to ping people when I do updates to the english version.　　

I have two ideas:
- Open a channel and add bots to ping and some routines. (I think `discord` is good)
- Forking another project to do Chinese version and adding the reference to the `README.md`  on `master`. Let volunteers(like me) to do the left things.

readlnh, 2020-01-14:
> > This is rather difficult I guess. I can try to ping people when I do updates to the english version.
> 
> I have two ideas:
> 
>     * Open a channel and add bots to ping and do some routines. (I think `discord` is good)
> 
>     * Forking another project to do Chinese version and adding the reference to the `README.md`  on `master`. Let volunteers(like me) to do the left things.
> 
>     * https://github.com/ColaPi/rust-raspi3-OS-tutorials

I can help review the translations in my free time.

andre-richter, 2020-01-14:
I think forking would reduce visibility quite some. How about having a working branch for Chinese, where you can maintain a README.cn.md through pull requests. I could then, from time to time, also sync finished versions to master as well.

readlnh, 2020-01-17:
> I think forking would reduce visibility quite some. How about having a working branch for Chinese, where you can maintain a README.cn.md through pull requests. I could then, from time to time, also sync finished versions to master as well.

I think it's a good idea, but I'm not sure if we can update the Chinese version as soon as master updates.

colachg, 2020-01-17:
#41 
@readlnh Now let's start.

andre-richter, 2020-01-17:
Sounds great! We can merge a PR whenever either @readlnh or @colachg has approved a PR of the respective other party. Since nobody in the WG can proof-read themselves, I think that's the straight forward way for now.

andre-richter, 2020-01-19:
Decision has been made and this activity has started now. Closing this issure :+1:

closed 2020-01-19 by andre-richter