rust-embedded/rust-raspberrypi-OS-tutorials #40: Using a RPi4 as the host

TimHutchinson, 2020-01-16:
Hi,
Have you ever used a RPi4 as a host for a RiP3 as the target. In this case for me the makefile does not build the kernel. It just seems to silently fail.

Everything works fine for me if I use an X64 machine as the host.

Thanks 

Regards
  Tim.

andre-richter, 2020-01-16:
Ha! That's an interesting use-case I never would have thought of :sunglasses: 

Absolutely no output at all when doing `make`?

TimHutchinson, 2020-01-16:
Hi,

Thanks for the quick reply. Below is the output from running make on one of
your tutorials. The copy towards the end fails because there is no kernel
built for the make file to copy. The kernel.d file however is written out.
I started out thinking the kernel was being built but put in some other
place, but I now don't think that is so.

As I said previously if I try this on my x64 Linux machine it all works
fine.

Regards

   Tim

------------------------------------------------------------------------
pi@raspberrypi:~/sources/rust-raspi3-OS-tutorials-master/02_runtime_init $
make
rm -rf target
RUSTFLAGS="-C link-arg=-Tsrc/bsp/rpi/link.ld -C target-cpu=cortex-a53 -D
warnings -D missing_docs" cargo xrustc
--target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
    Updating crates.io index
   Compiling core v0.0.0
(/home/pi/.rustup/toolchains/nightly-2019-12-20-armv7-unknown-linux-gnueabihf/lib/rustlib/src/rust/src/libcore)
   Compiling compiler_builtins v0.1.24
   Compiling rustc-std-workspace-core v1.99.0
(/home/pi/.rustup/toolchains/nightly-2019-12-20-armv7-unknown-linux-gnueabihf/lib/rustlib/src/rust/src/tools/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/tmp/xargo.npQWs8qhTaHC)
    Finished release [optimized] target(s) in 1m 21s
   Compiling kernel v0.1.0
(/home/pi/sources/rust-raspi3-OS-tutorials-master/02_runtime_init)
    Finished release [optimized] target(s) in 0.73s
cp target/aarch64-unknown-none-softfloat/release/kernel .
cp: cannot stat 'target/aarch64-unknown-none-softfloat/release/kernel': No
such file or directory
make: *** [Makefile:60: kernel8.img] Error 1

andre-richter, 2020-01-16:
Okay, so something is getting successfully compiled it seems. Maybe the name or path that the Makefile assumes differs.

Can you take a look into the target folder?

TimHutchinson, 2020-01-16:
As requested here are the contents of the target directory

pi@raspberrypi:~/sources/rust-raspi3-OS-tutorials-master/02_runtime_init/target
$ ls -R
.:
aarch64-unknown-none-softfloat  release

./aarch64-unknown-none-softfloat:
release

./aarch64-unknown-none-softfloat/release:
build  deps  examples  incremental  kernel.d

./aarch64-unknown-none-softfloat/release/build:

./aarch64-unknown-none-softfloat/release/deps:
kernel-dee76b2943f6c6b2.d

./aarch64-unknown-none-softfloat/release/examples:

./aarch64-unknown-none-softfloat/release/incremental:

./release:
build  deps  examples  incremental

./release/build:

./release/deps:

./release/examples:

./release/incremental:

andre-richter, 2020-01-17:
Hmm, strange... I don't have an explanation for that right now from the top of my head.

TimHutchinson, 2020-01-17:
Me neither, but thanks for looking.

I added the verbose flag to the xrustc command in the makefile but it
didn't help. There is one thing I did different from your tutorial. I
already had a Rust setup on my RPi4 so I just installed the nightly build
and required components by hand and switched the cargo default toolchain.
Maybe I missed something.

I'll keep poking around. I have a spare SD card so I might go back and
start again exactly following your curl script. (I did this on the x64
machine). I'll let you know if I find anything.

BTW thank you for doing this project.

  Tim.

andre-richter, 2020-02-11:
Hi Tim,

any updates on this? I think I would close this issue soon otherwise until further questions pop up.

Thanks,
Andre

TimHutchinson, 2020-02-11:
Hi Andre,

No joy I'm afraid. I tried various things including running everything as
root in case it was some weird permissions problem. I even did a fresh
install and exactly followed your instructions but still it would not build
the kernel. Recently I installed a 64 bit version of Ubuntu on my Pi4 and
tried to build your code there, but still the same problem it just silently
fails. I think it might be an error in the linker as then maybe any error
message is just getting thrown away by cargo. But that is just a guess

I don't think I'm going to make any further progress.So will switch to
using an x86 machine where it all seems to work ok.

You can close the log.

Thanks for your time.

Regards

  Tim.

andre-richter, 2020-02-11:
Alright, thanks for the update and sorry that I currently lack bandwidth to dig into that myself.

Best,
Andre

closed 2020-02-11 by andre-richter