rust-embedded/rust-raspberrypi-OS-tutorials #42: How to get start?

colachg, 2020-01-19:
My environment is Ubuntu 18.04.3 LTS with docker-ce 19.03.2. I follow the `README.md` and have install the suitable Rust toolchain. 
When I run `make qemu`, get this output:
```
rm -rf target
RUSTFLAGS="-C link-arg=-Tsrc/bsp/rpi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo xrustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
error: The sysroot can't be built for the Stable channel. Switch to nightly.
note: run with `RUST_BACKTRACE=1` for a backtrace
Makefile:56: recipe for target 'target/aarch64-unknown-none-softfloat/release/kernel' failed
make: *** [target/aarch64-unknown-none-softfloat/release/kernel] Error 1
```
I checked the Makefile and don't know what to do.

andre-richter, 2020-01-19:
You havent installed or selected _nightly_ toolchain.

colachg, 2020-01-19:
Thanks.I am newbie to Rust. After some search I make it worked with below commands:
`rustup toolchain install nightly-2019-12-20;rustup  default nightly-2019-12-20-x86_64-unknown-linux-gnu;rustup component add rust-src`
:)

closed 2020-01-19 by colachg