rust-embedded/rust-raspberrypi-OS-tutorials #43: Arduino emulator on bare metal PI.

cheako, 2020-01-24:
I found this project in Google and am asserting that here is a good a place as any for this post.

I had an interesting discussion on https://youtu.be/cWuFLEb-pUM
The idea there was that since an Arduino does less it has some advantage over the less expensive PI zero W($20 cheeper).  My point was that it would be entirely possible to have a PI do less.  This repo very effectively demonstrates this, I think.

The challenge is to write something that looks like an Aurduino, bit is a PI running a not so fancy OS.

theAkito, 2020-01-25:
Great idea! Was looking for something like this for my next self-made keyboard.

andre-richter, 2020-01-25:
I am not sure if I understand the question entirely. If you are saying that you want to program the Pi like an Arduino, that is, have a single main loop or state machine in a single threaded application running on the pi, then yes, you can use the tutorial code here as a template.

Basically, put your state machine into kernel_main().

andre-richter, 2020-02-11:
This issue has been dormant for some time. I am closing it right now. Feel free to reopen if questions remain.

closed 2020-02-11 by andre-richter