rust-embedded/rust-raspberrypi-OS-tutorials #46: Debug using Serial Debug Cable

vinaychandra, 2020-02-16:
Is it possible to debug using the Serial Debug Cable rather than using a dedicated JTAG?

rahealy, 2020-02-17:
Yes. https://github.com/rust-embedded/rust-raspi3-OS-tutorials#-usb-serial-output

vinaychandra, 2020-02-17:
That is useful to do booting via USB. My question was regarding debugging using GDB.

rahealy, 2020-02-17:
AFAIK the RPi's JTAG is implemented at the transistor level in the hardware. As a result JTAG requires a JTAG adapter - either built from plans found on the internet and/or hacked together and/or purchased from a dealer like Olimex.

You also might consider emulation using the qemu target in the tutorial's Makefile. Please see this comment https://github.com/rust-embedded/rust-raspi3-OS-tutorials/issues/45#issuecomment-584795461 for an example of how to modify the _12_cpu_exceptions_part1_ tutorial to connect an instance of `gdb-multiarch` to qemu running in the docker image provided by the tutorial's author @andre-richter.

rahealy, 2020-02-17:
@vinaychandra  Forgot to mention: The reason I answered yes to your original question is that your programs can read/write to the RPi UART and send debug info over the serial->USB converter to a computer running a terminal emulator like PuTTY.

vinaychandra, 2020-02-17:
Thanks for the info! Yes, while I do understand that we can send random debug info over UART, I am looking more for gdb support over USB so that breakpoints and watches would work. I also do understand that qemu is an option, but I wanted to see if it is possible to do breakpoint debugging over USB/UART

rahealy, 2020-02-17:
@vinaychandra I did some research. The _Arm® Architecture Reference Manual
Armv8, for Armv8-A architecture profile_ chapter D2 _AArch64 Self-hosted Debug_  describes the hardware debugging goodies for the ARM cores on the RPi3 SoC. So it might be possible. Has anyone written a crate that you can just add as a dependency and begin debugging? I don't know.

andre-richter, 2020-02-17:
As @rahealy already pointed out, this is currently not possible.

While it would be possible to come up with a software solution, e.g. implementing a GDB Remote Serial Protocol server in an EL > EL1, this is unfortunately out of the scope for now.

It would be a fun project, but given that JTAG adapters are quite affordable, not worth pursuing in light of the effort that would be needed.

andre-richter, 2020-02-28:
Closing for now since this feature is not on the agenda for the near future.

closed 2020-02-28 by andre-richter