rust-embedded/rust-raspberrypi-OS-tutorials #56: How can chainloader jump to the new location?

FedericoPonzi, 2020-05-18:
Hello,
In part 7: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/07_uart_chainloader
I cannot understand how the kernel can be correctly loaded, even if you set this in the linker file:
```
-    /* Set current address to the value from which the RPi starts execution */
-    . = 0x80000;
+    /* Set the link address to 32 MiB */
+    . = 0x2000000;

```
(Correct me if I'm mistaken) when you set `.=0x2000000` the kernel will be loaded at that position in memory.
Then as soon as GPU initialization is done, the cpu will start running instruction at location 80_000. But how can this works if the kernel is in position 32Mib?
I know we could overwrite the kernel start location by putting `kernel_address=0x4000000` inside the config.txt, but since this wasn't requested in the tutorial, I didn't do it but still it's working and also nicely verifiable via `qemuasm`.
How is this possible?
Thanks!

andre-richter, 2020-05-18:
The key part is that the GPU will load the binary from the SD card to 80_000 no matter what. And it will start executing code from there.

The chain loader binary, therefore, will run its first instructions from 0x80_000, despite having been linked to 0x2000000.

That is not an issue though, because we compiled the chain loader binary as position independent code. One of the first things the chain loader does is to then copy itself from the 0x80_000 address to the 32 MiB address, and then continue execution from there. This way we can safely put the binary that comes in over UART at the standard 0x80_000 because we do not execute from there anymore and as such do not overwrite the chain loader binary.

I’m summary, the load address in the linker script is in this case actually only used to retrieve where to relocate the chain loader during runtime.

Hope that helps.

FedericoPonzi, 2020-05-20:
Thanks for the super fast answer, and sorry for my super late reply! Indeed it was very useful, in my noobness I didn't know about the PIC option.

closed 2020-05-20 by FedericoPonzi