rust-embedded/rust-raspberrypi-OS-tutorials #57: Failing building the kernel

aurelien-m, 2020-05-30:
Hey,

I'm having the following problem when I try to execute `make qemu`:
```
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
    Finished release [optimized] target(s) in 0.07s
Failed to execute tool: objcopy
No such file or directory (os error 2)
Makefile:62: recipe for target 'kernel8.img' failed
make: *** [kernel8.img] Error 101
```

I have no clue of what is wrong, I tried looking it up but can't find a fix. The `objcopy` command also works fine in the terminal. I'm running the following configuration by the way:
```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/aurelien/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
nightly-2019-12-20-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu (default)

installed targets for active toolchain
--------------------------------------

aarch64-unknown-linux-gnu
aarch64-unknown-none-softfloat
x86_64-unknown-linux-gnu

active toolchain
----------------

nightly-x86_64-unknown-linux-gnu (default)
rustc 1.45.0-nightly (4bd32c980 2020-05-29)
```
Do I need to do something with the docker files? Thanks in advance :)

andre-richter, 2020-05-30:
From the error message I can at the moment not conclude what exactly is missing.

If it is the rust-objcopy you may have missed this step:

cargo install cargo-binutils

aurelien-m, 2020-05-30:
I have already installed the `cargo-binutils`:
```
    Updating crates.io index
     Ignored package `cargo-binutils v0.3.0` is already installed, use --force to override
```

The strange thing is that the `rust-objcopy` commands gives me this error in the terminal:
```
Failed to execute tool: objcopy
No such file or directory (os error 2)
```
But `objcopy` works just fine.

andre-richter, 2020-05-30:
Hmm maybe just  something with your rustup state went bogus.

Can you try a rm -rf ~/.rustup ~/.cargo and install from scratch?

newAM, 2020-05-30:
I have the same problem.

```
rustup show           
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/alex/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
nightly-2020-05-04-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu (default)

installed targets for active toolchain
--------------------------------------

aarch64-unknown-none-softfloat
x86_64-unknown-linux-gnu

active toolchain
----------------

nightly-x86_64-unknown-linux-gnu (default)
rustc 1.45.0-nightly (4bd32c980 2020-05-29)
```

I did a `--force` with my install.
```
cargo install cargo-binutils --force
```

The `make qemu` target in `01_wait_forever` fails for me:

```
make qemu                           
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
    Finished release [optimized] target(s) in 0.00s
Failed to execute tool: objcopy
No such file or directory (os error 2)
make: *** [Makefile:62: kernel8.img] Error 101
```

newAM, 2020-05-30:
Found the source of my problem.

I went to the cargo-binutils [README](https://github.com/rust-embedded/cargo-binutils#installation), and they have an added step for install:

```
$ cargo install cargo-binutils

$ rustup component add llvm-tools-preview
```

I ran `rustup component add llvm-tools-preview` and it works for me now.  Does this solve your problem @aurelien-m ?

aurelien-m, 2020-05-30:
I think the `llvm-tools-preview` component is added during this command:
```
curl https://sh.rustup.rs -sSf | sh -s --  \
    --default-toolchain nightly-2020-05-04 \
    --component llvm-tools-preview rustfmt
```

However, I did a full reinstall with `rustup self uninstall`, did all of the commands and now I get this:
```
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
/bin/sh: 1: cargo: not found
Makefile:59: recipe for target 'target/aarch64-unknown-none-softfloat/release/kernel' failed
make: *** [target/aarch64-unknown-none-softfloat/release/kernel] Error 127
```

The `kernel8.img` has still been built though.

newAM, 2020-05-30:
Well, I did not read the instructions carefully enough, whoops!

This error: `/bin/sh: 1: cargo: not found` looks like it is not in the path, did you run `source $HOME/.cargo/env` after the reinstall?

aurelien-m, 2020-05-30:
Yep, I've done that, I can launch the `cargo` command perfectly fine in my terminal.

andre-richter, 2020-05-30:
I’ll try a from-scratch install later tonight with the Instructions I’ve put on the main repo readme and see if I hit this too.

I didn’t expect any regressions since the CI nightlys build fine.

andre-richter, 2020-05-30:
Hi,

these steps worked for me:

1. Open Terminal.
2. rm -rf ~/.rustup ~/.cargo
3. Close and open Terminal again.
4. curl https://sh.rustup.rs -sSf | sh -s --  --default-toolchain nightly-2020-05-04 --component llvm-tools-preview rustfmt
5. Close and open Terminal again.
6. rustup target add aarch64-unknown-none-softfloat
7. cargo install cargo-binutils

```console
01_wait_forever [master] $ make
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
    Finished release [optimized] target(s) in 0.10s
```

This is basically the top level README with some additional terminal close/open mumbo jumbo to be extra sure the ENV is in correct state.

In summary, it works for me and @newAM locally, and it works on the CI nightly builds. Strong indications that it is something local on your machine.

aurelien-m, 2020-05-30:
Thanks for the help, I did exactly that, and the `make` works perfectly. However when I do a `make qemu` I get this:
```
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
    Finished release [optimized] target(s) in 0.06s
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
Makefile:72: recipe for target 'qemu' failed
make: *** [qemu] Error 126
```

What I tried is `sudo make qemu` which brings me to the error I had before:
```
RUSTFLAGS="-C link-arg=-Tsrc/bsp/raspberrypi/link.ld -C target-cpu=cortex-a53 -D warnings -D missing_docs" cargo rustc --target=aarch64-unknown-none-softfloat --features bsp_rpi3 --release
/bin/sh: 1: cargo: not found
Makefile:59: recipe for target 'target/aarch64-unknown-none-softfloat/release/kernel' failed
make: *** [target/aarch64-unknown-none-softfloat/release/kernel] Error 127
```

**Finally**, what works for me is changing the Makefile and adding `sudo` right here and launching it with `make qemu`:
```
DOCKER_CMD           = sudo docker run -it --rm -v $(shell pwd):/work/tutorial -w /work/tutorial
```

This is kind of strange, if you have time to check, what version of docker do you use? Hope this will maybe also help others.

closed 2020-05-30 by aurelien-m

andre-richter, 2020-05-30:
Maybe your user is not in the docker group: https://docs.docker.com/engine/install/linux-postinstall/

bharatgohil, 2021-12-13:
> Found the source of my problem.
>
> I went to the cargo-binutils [README](https://github.com/rust-embedded/cargo-binutils#installation), and they have an added step for install:
>
> ```
> $ cargo install [...]

thank you. I have same problem. fixed with this solution.

HamPoole, 2022-06-26:
Just commenting.

I also ran into this issue, and also solved it via the same hack Aurelien used, adding a sudo to the Makefile.
```
me@pop-os:~/projects/rust-raspberrypi-OS-tutorials/01_wait_forever$ docker run hello-world

Hello from Docker!
This message shows that your installation appears to be working correctly.

```

Unless I'm completely wrong, I'm fairly sure I've added the docker usergroup correctly, and applied it to my Linux user.

Very perplexed about the root cause of this issue.