rust-embedded/rust-raspberrypi-OS-tutorials #60: remove deprecated --path flag from Readme

Am3ra, 2020-06-02:
When following the install scripts in the main README, i get the following warning:

bundle install --path .vendor/bundle

[DEPRECATED] The `--path` flag is deprecated because it relies on being remembered across bundler invocations, which bundler will no longer do in future versions. Instead please use `bundle config set path '.vendor/bundle'`, and stop using this flag

andre-richter, 2020-06-03:
I had previously tried the suggested alternative command, but it did not work for me. Hence I haven't addressed this yet. 

Did you give it a try?

Am3ra, 2020-06-11:
No, but thanks for the suggestion, didn't even consider it! I will! (Btw, love this repo, still trying to figure out how to use it well)

Am3ra, 2020-06-11:
Completely uninstalled the gem, and reinstalled it without the `--path` flag, installed correctly and then ran the `bundle config set path '.vendor/bundle'` command. No problems so far running.

andre-richter, 2020-06-16:
> Completely uninstalled the gem, and reinstalled it without the `--path` flag, installed correctly and then ran the `bundle config set path '.vendor/bundle'` command. No problems so far running.

I think that only worked because it installed the Gems globally :thinking: 

The goal is that the installation of the gems itself ends up in the vendor folder, so installation would need to happen after `bundle config set path '.vendor/bundle'` but it then never worked for me like intended.

I'll keep the bug open until I find some motivation to revisit this :smile:

commit 0586a7b3 (2020-06-29): Use bundle config instead of deprecated --path

Closes #60
commit 44beff8b (2020-06-29): Use bundle config instead of deprecated --path

Closes #60
closed 2020-06-29 by andre-richter