rust-embedded/rust-raspberrypi-OS-tutorials #64: Suggestion: Instructions for people who already have a Rust toolchain

noneuclideangirl, 2020-06-24:
Hi there,

Thanks for this effort -- it's amazing!
I'd just like to make a quick suggestion. It might be nice to add instructions for people who already have Rust installed, like myself. I had a lot of trouble getting started before I finally realised I needed to run `rustup component add llvm-tools-preview`. I think this, plus instructions to switch to `nightly`, would make a nice addition to "The tl;dr Version" of System Requirements for other people like myself.

Happy to make a PR if this sounds good.

andre-richter, 2020-06-24:
Thanks, glad you like it.

I think I understand the problem.
I don’t have access to a PC for around a week, I’ll address this once I have.

Just curious, what happens if you do the steps in the current TLDR. 
Would the rustup not install the respective nightly and the llvm tools alongside an existing install?

Best,
Andre

eleanor-em, 2020-06-26:
Hey Andre, good question -- I did not touch the instructions for fear of breaking my existing install, since I was greeted with the usual installation screen when I ran the rustup script. However it seems you are right and that it would have been harmless.

Unfortunately I can't test whether it correctly installs the LLVM tools but I imagine it would. Perhaps it's worth a comment reassuring users that it's safe to do this if you already have Rust installed.

commit 1b91a484 (2020-06-28): Add instructions for preexisting Rust installation

Closes #64
commit b0b9650d (2020-06-28): Add instructions for preexisting Rust installation

Closes #64
commit b5fd8545 (2020-06-28): Add instructions for preexisting Rust installation

Closes #64
closed 2020-06-28 by andre-richter