rust-embedded/rust-raspberrypi-OS-tutorials #65: Is the change in 04_zero_overhead_abstraction valid?

imsut, 2020-06-25:
First of all, thank you for organizing this material.  This is of great help for me to learn Rust.

I have a question about the change in "04_zero_overhead_abstraction" where assembly code is replaced with Rust.
In [`_start` function](https://github.com/imsut/rust-raspberrypi-OS-tutorials/blob/master/04_zero_overhead_abstraction/src/_arch/aarch64/cpu.rs#L27-L28), there are at least two function calls (`cpu::smp::core_id` and `SP.set`) before the stack pointer is initialized.  For this code to work, the Rust compiler has to inline these function calls.
The `inline` attribute, however, is just a hint and, in my understanding, the Rust compiler can ignore the hint.
https://doc.rust-lang.org/reference/attributes/codegen.html#the-inline-attribute

I feel like anything before the stack pointer initialization needs to be written in assembly to avoid function calls.
Curious to hear your thoughts.

alilee, 2020-06-26:
This is a very interesting question, but also remember small functions may not use the stack. 

This does come up - if I compile with opt-level=0 then I think my code hangs for this reason.

andre-richter, 2020-06-26:
You‘re right.

Setting the stack pointer should ideally be done in early boot assembly before the first function call happens.
You cannot rule out that the compiler does not honor the inline attributes in this code snippet.

I threw out the assembly here nevertheless, mostly “because we can”, because I like to have as much Rust code as possible in this toy-kernel 😎.
We have emulation for testing and a corresponding CI pipeline that would raise a flag if the compiler inserts function calls here, because the booting would fail early on.
Also, for functions that boil down to very few lines of assembly (in the SP case a single one), you can be quite optimistic that the compiler honors the inline suggestion.

In the [original version of this particular tutorial](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/original_version/07_abstraction#boot-code) I mentioned that we would need to manually  check that function calls don’t happen here.
I wanted to mention it in the new version here too,  it didn’t come around porting the text over yet.

Hope that helps!

Best,
Andre

imsut, 2020-06-26:
@andre-richter -- thx for the explanation and the pointer to the original post!

> mostly “because we can”, because I like to have as much Rust code as possible in this toy-kernel sunglasses.

makes sense :)

closed 2020-06-26 by imsut