rust-embedded/rust-raspberrypi-OS-tutorials #69: __bss_start and __bss_end are empty

FedericoPonzi, 2020-07-12:
Hello!
In the step 05_safe_globals, I've tried to add the following code in the runtime_init/bss_range function:
```
26 |       panic!("Bss start:{:?}, end:{:?}", __bss_start, __bss_end);
```

This is the resulting output:

```
Kernel panic: Bss start:26, end:0
````
I've tried the same in step 09_timestamps:
```
Kernel panic: Bss start:0, end:0
```
Any clues on what's going on or how to fix this? Thanks!

andre-richter, 2020-07-12:
Hi,

I think what you actually wanted is 
```rust
panic!(
    "Bss start:{:?}, end:{:?}",
    &__bss_start as *const usize, &__bss_end as *const usize
);
```
You were reading the _contents_ at the symbol address (the memory there is probably uninitialized and can therefore have random content).

In your case, you want the symbol's _address/location_.
That is one of the more confusing parts everybody learning OS dev will once stumble upon.

Some resources:
https://en.wikipedia.org/wiki/Symbol_table
https://wiki.osdev.org/Linker_Scripts

By the way, for looking at the `bss` symbols, the Makefiles provide a handy command already:
```Makefile
make nm
```

Best,
Andre

FedericoPonzi, 2020-07-12:
Thank you very much for your fast and insightful answer! You were right, and thanks for pointing me to nm of which I wasn't aware of.

closed 2020-07-12 by FedericoPonzi