rust-embedded/rust-raspberrypi-OS-tutorials #70: UART Baud Rate calculation

cfsamson, 2020-07-13:
Let me first thank you for this fantastic tutorial. I'm having loads of fun going through it all and the code is just beautifully organized!

I have some problems on chapter 6 (which, by excluding everything else is probably due to a broken SD card) and I had to go through every line of code and double check everything.

This is nitpicking since the result is still within ~5 % error range of the tx/rx channel which should be acceptable anyway but since this is a tutorial it might still be worth bringing up:

in `bcm20xx_pl011_uart.rs` we have the `PL011UartInner::init` method which sets up UART.

The calculations for the baud rate divisor given a target baud rate of `230400` is as follows: `(48_000_000/16)/230400 = 13,02083`. The integer part is `13` is written to the `IBRD` bitfield. However, the fractional part is written to a the 6-bit field `FBRD`. Now, as I understand it, since the fractional field can only hold values from 0-63, the right thing to do is to divide the decimal fractional on 64: `0,0208*64 = 1,3312` which gives a "nearest value" of 1 instead of 2.

As you might understand I'm on the learning path here so I'm wary of making too bald claims about the correctness but I spent quite some time to check how you got the numbers you got and I leaned on [this resource - 11.2.2. TM4C UART Details](http://users.ece.utexas.edu/~valvano/Volume1/E-Book/C11_SerialInterface.htm) since I wondered what would be the right thing to do if the fractional was a value higher than 63.

I know this is very nit picky but it would be interesting to add at least the basic calculations in the comments for the code block as well since it's not easy to understand and it really does connect some dots that are hard to find otherwise.

Again, thanks for creating this great resource.

andre-richter, 2020-07-13:
Hi @cfsamson,

glad to hear that you like the tutorials :)

Regarding the baud rate calculation, I have to confess that I probably haven't looked as deep into it as I should have. I remember getting frustrated that I was not able to find a good all-in-one explanation of how to do them (the respective Broadcom datasheet is bad), so I ended up reading other source-code and drew conclusions from what I saw there.
It then happened to work and I was happy that I don't need to look into it anymore as I found it to be one of the less fun things (it's important still, no doubt).

So yes, I agree that 1 seems to be the correct value. I'll prepare a patch to fix this!
For lack of time, I won't update the tutorial text due to lack of spare time at the moment. If you feel like it, I'm happy to take a PR explaining this part :)

Thanks,
Andre

closed 2020-07-13 by andre-richter

cfsamson, 2020-07-13:
That's great. I'll make create a PR soon.

commit 87e74f62 (2020-07-13): UART: Fix FBRD

Closes #70