rust-embedded/rust-raspberrypi-OS-tutorials #78: While following the tutorial I get "can't find crate for `test`" while compiling for unit test execution

2ndTaleStudio, 2020-10-04:
Hi there,

I'm using a custom build target for my Raspberry Pi bare metal kernel. After setting everything up and compiling the binary kernel for unit testing I get the compiler error "can't find crate for `test`".

Any clue what might be the root cause for this?

I'm not using cargo xbuild, but the new cross compile configuration in /.cargo/config like so:
```toml
[unstable]
build-std = ["core", "compiler_builtins", "alloc"]
```

Any help would be much appreciated.

Thanks in advance.

andre-richter, 2020-10-04:
Hi,

specific to my tutorial, you would get this whenever rustc comes across a `#[test]` and not a `#[test_case]`  as it is emitted by the macro (https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/13_integrated_testing/test-macros/src/lib.rs#L22).
I had this case while writing the tutorial and couldn't figure out whats going on. Turned out a third-party dependency that I pulled in through cargo had a macro that not only emitted code, but also a unit-test with it (it was the tock-registers). That was very hard to find.

I can't tell anything about building `std` with cargo, since I haven't touched that yet. Until now, just using the precompiled std for `aarch64-unknown-none-softfloat` is sufficient.

andre-richter, 2020-11-14:
Closing this for now since no action on the issue.

Please reopen for follow-up questions.

closed 2020-11-14 by andre-richter