rust-embedded/rust-raspberrypi-OS-tutorials #80: Examples 6 and 7 not working in RPI4B

CPerezz, 2020-11-07:
Hey. Firs of all, thanks for this great tutorial. Is really fun & interesting!!

As for the problems I'm experiencing:

## Example 6
- Ran `BSP=rpi4 make` and copied `kernel8.img` together with `fixup4.dat`,  `start4.elf` (renamed to `start.elf` as per README comment) and `bcm2711-rpi-4-b.dtb` and `config.txt` with the specified contents to the SD.
- Ran `sudo screen /dev/ttyUSB0 230400`. Black screen appears and then after pressing `Enter` nothing happens.

I tried with `start4.elf` and doesn't work. Also tried to download the latest firmware and same outcome.. 

## Example 7
- Ran `BSP=rpi4 make` and copied the `kernel8.img` to the SD. 
- Ran `BSP=rpi4 make chainboot` and this was the outcome:
```
BSP=rpi4 make chainboot
Minipush 1.0

[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
^C
[MP] Bye 👋

```

The error is very similar to what I've seen in #79 . I checked whether I had everything correct with the toolchains and targets and the rest of the config. But since I've seen new commits have been done since this issue was close, maybe one of them broke this.

If you need any kind of extra info, just ping me. Sorry that I cannot provide more, but I don't really know how to debug it.

andre-richter, 2020-11-08:
Hi @CPerezz,

that's strange. I just re-tested it on my Pi4. I loaded the newest versions of `start4.elf`, `fixup4` and the `dtb`.
It works for me, and it doesn't make a difference if the start is named `start.elf` or `start4.elf`.

- Do you have any other files on SD card?
- Are you ejecting the SD card properly after writing to it from the host? This can often cause issues because the contents are not flushed onto the SD but reside in the OS cache when you don't eject/flush properly.

-

CPerezz, 2020-11-08:
Hi @andre-richter , thanks for answering.

No other files are there:
```
> ls -la /run/media/boot/$USER
Permissions Size User Date Modified Name
.rw-r--r--   47k kr0   7 Nov 21:22  bcm2711-rpi-4-b.dtb
.rw-r--r--    56 kr0   8 Nov  0:26  config.txt
.rw-r--r--  5.4k kr0   7 Nov 21:22  fixup4.dat
.rw-r--r--  8.2k kr0   8 Nov  0:20  kernel8.img
.rw-r--r--  2.2M kr0   7 Nov 21:22  start.elf
```

Also. After writing/copying files to the SD I just do `sudo umount /dev/sdYX` and eject the USB adapter that has the SD card in it.

Will try to re-do completely from scratch example 6 and 7 later today to see what happens. But at this point. I tried everything. So if I can provide any other info that might help the debugging. Just tell me. Since it's weird, and also difficult to check since `qemu` doesn't support rpi4 on the -M option.

CPerezz, 2020-11-08:
kk after doing everything from scratch again:

### Example 6
I probably did something wrong since now I'm capable of booting the RPI4B and the `Enter` triggers the execution as expected breaking the infinite loop. Although the prints are indented and so the output looks like:
```
Output 1
               Output 2
                               Output 3
                                               Output 4
```
But it's fine. I played a bit with the println! macro to try to see why this happens but haven't been able to catch it yet.

### Example 7
The kernel boots correctly if I load it in the SD and use `screen` as in example 6.

But when I run `BSP=4 make chainboot` i keep getting the same errors:
```
Minipush 1.0

[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again

[MP] ⏳ Waiting for /dev/ttyUSB0
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again

^C
[MP] Bye 👋
^C
make: *** [Makefile:104: chainboot] Error 
```

andre-richter, 2020-11-08:
Hi @CPerezz,

I was able to reproduce the odd line breaks in tutorial 6. A while ago, I changed all the kernel sources to conclude a line only with `\n`, and not print an explicit carriage return. The terminal part of `minipush` accounted for it, but I never realized that it breaks `screen`.

I carved out the terminal part of minipush for tutorial 6, so connection to the RPi now happens through an in-house tool as well. Please check out if the changes introduced in https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/commit/e1473099ff0b4ffc4c07fa24985bb55d8ef28f07 work for you.

Regarding tutorial 7, hard to say why this still fails. Even more so because tutorial 6 seems to work...

abdes, 2020-11-09:
@andre-richter thanks again for doing these tutorials. Maybe this is related to my issue #81 ?

andre-richter, 2020-11-09:
@CPerezz, 

can you check if something changes when you use [this kernel8.img](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/files/5513557/kernel8.zip) as the chainloader on your Pi4 SD card?

It has the changes @abdes reported in #81.

andre-richter, 2020-12-09:
Closing this for now since no feedback received.

We can reopen anytime if needed.

closed 2020-12-09 by andre-richter

CPerezz, 2020-12-11:
Hi @andre-richter I've been extremely busy. That's why I haven't been able to test your changes. I'll do ASAP. Sorry for the delay.