rust-embedded/rust-raspberrypi-OS-tutorials #85: QUESTION: in 06_drivers_gpio_uart, Why PanicUart is initialized after map_pl011_uart in  fn panic_console_out?

umaYnit, 2020-12-04:
I am new to Raspberry Pi and operating system, so I don’t know much about circuits and software.
In  `main.rs` file, as the comment says, the fn `post_device_driver_init` is executed after device_drivers `init`.but method `panic_console_out` in `bsp\raspberrypi\console`, it's reverse. Hope to get your answer

andre-richter, 2020-12-06:
Hi,

I am not sure I understand. In `panic_console_out` in tutorial 6, we don't call `post_device_driver_init` ?

umaYnit, 2020-12-07:
In `main.rs\kernel_init()`, `post_device_driver_init` called after `all_device_drivers` is `init()`, and the method  `post_device_driver_init` just called `super::GPIO.map_pl011_uart();` , finally it is `bcm2xxx_gpio.rs\map_pl011_uart(_)`.
And in `bsp\raspberrypi\console\panic_console_out()`, the  `bcm2xxx_gpio.rs\map_pl011_uart(_)` called before `panic_uart.init();`.
Is it because their order is not important ?

closed 2020-12-09 by andre-richter