rust-embedded/rust-raspberrypi-OS-tutorials #86: [07_uart_chainloader] Make chainboot only works if I power my RPi3 after status `[MP] ✅ Connected`

tobiassoppa, 2020-12-06:
Hi,

I encountered quite a weird issue in 07_uart_chainloader. After starting `make chainboot`, I **never** got the desired output 
```
[MP] ✅ Connected
 __  __ _      _ _                 _ 
|  \/  (_)_ _ (_) |   ___  __ _ __| |
| |\/| | | ' \| | |__/ _ \/ _` / _` |
|_|  |_|_|_||_|_|____\___/\__,_\__,_|

           Raspberry Pi 3            

[ML] Requesting binary
[MP] ⏩ Pushing 6 KiB ==========================================🦀 100% 0 KiB/s Time: 00:00:00
[ML] Loaded! Executing the payload now

[0] Booting on: Raspberry Pi 3
[1] Drivers loaded:
      1. BCM GPIO
      2. BCM PL011 UART
[2] Chars written: 93
[3] Echoing input now
```
Instead, I frequently got one of the following outputs:
```
[MP] ✅ Connected

[MP] ⚡ Protocol Error: Remove and insert the USB serial again
```

```
[MP] ✅ Connected
?? $Error ? ?$Error ?
[MP] ⚡ Protocol Error: Remove and insert the USB serial again
```

```
[MP] ✅ Connected
ting the payload now

[ML] Loaded! Executing the payload now

[ML] Loaded! Executing the payload now

[ML] Loaded! Executing the payload now

<repeats until I stop>

[MP] Bye 👋
```

```
[MP] ✅ Connected
?????????????????????????????????????????????????????????????????????????????????????
?????????????????????????????????????????????????????????????????????????????????????
?????????????????????????????????????????????????????????????????????????????????????

<repeats until I stop>

[MP] Bye 👋
```

After several hours I finally figured out how to work around the issue:
* If I power my RPi3 like described in the [wiring diagram](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/doc/wiring.png), I **always** get the issues I described above
* If I power my RPi3 via micro-usb, I mostly get the issues described above, but **sometimes** it works. 

It **always** works if I do it like this:
1. `make chainboot` 
2. Plug in my [USB serial cable](https://www.amazon.de/gp/product/B01N4X3BJB/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) **without** using the red +5V wire (I linked the cable I use, in case this is important)
3. Wait until `[MP] ✅ Connected` gets displayed
4. Use micro-usb to power the RPi3.

I don't really understand the issue in detail, but to me it seems like it's somehow connected to timings/delays. The only place in the code I could find that directly influences said timings/delays is `const DELAY: usize = 2000` in [src/bsp/device_driver/bcm/bcm2xxx_gpio.rs](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/07_uart_chainloader/src/bsp/device_driver/bcm/bcm2xxx_gpio.rs#L152). I tried some values here and if I set it to `const DELAY: usize = 20000`, it also works **sometimes** (but at this point I have no clue if the change really reduces my error rate, or if it just happened by chance and led me to wrong conclusions).

Do you have any idea what really causes the error? If so, I would appreciate if you could provide a little write-up what is happening on the lower levels and what causes the error I got. Since chasing this down took more than half of my day but I still didn't really understand it, I'm quite curious now 😄

andre-richter, 2020-12-06:
Hi, sorry for the inconvenience!

Unfortunately, it is hard to exactly tell where it goes wrong.

Due to the fact that it always works when you power the RaspberryPi through Micro-USB, the best bet is that your USB-Serial cable is not able to provide enough power to the RaspberryPi. In this case, anything can happen.
If it is not the serial cable that is the bottleneck, maybe its the USB port. You could try different USB ports if that changes something.

Anyways, your message makes me realize that with different people using different serial cables and computers, assuming that powering over the serial works for everyone doesn't scale. It is convenient, that's why I went for it originally, but it was clear that underpowering can be an issue.

I'll take an action item to change the tutorials so that they demand the Pi being powered with an officially supported external power supply. This will also require changes to the chainloader protocol.
But please don't wait for me, I hope to get it done during the upcoming holidays.

FYI, delays on the host-side are in the Minipush and Miniterm scripts (but I don't think they are the issue here):
- https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/utils/miniterm.rb#L40
- https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/utils/minipush.rb#L30

tobiassoppa, 2020-12-06:
Thank you so much for the quick answer and insightful answer! I certainly did not consider power to be an issue (very new to low level programming / interacting with hardware, so lots to learn). 

Already checked the Scripts out earlier, because I wanted to understand the chainload protocol. So I tried to use CoolTerm and sending everything manually. Was quite interesting and certainly helped my understanding of what is going on, but didn't work in the end.

Since you mentioned it, enjoy your holidays! 😄

closed 2020-12-06 by tobiassoppa