rust-embedded/rust-raspberrypi-OS-tutorials #95: Unexpected Error: #<Errno::EINVAL: Invalid argument - tcsetattr>

sachaos, 2021-01-03:
I have a problem when trying 06_drivers_gpio_uart.
I got following error when running error.

```
$ DEV_SERIAL=/dev/tty.usbserial-1420 make miniterm

Miniterm 1.0

[MT] ⚡ Unexpected Error: #<Errno::EINVAL: Invalid argument - tcsetattr>

[MT] Bye 👋
```

I think the cause of the error is `baud`.
The baud is depending on the platform according to the documentation.
I'm using Mac OS.

> Integer from 50 to 256000, depending on platform)
> https://rubydoc.info/gems/serialport/SerialPort:set_modem_params

## Reproduce process

1. Use Mac OS.
1. Compile source.
1. Write binary to SD card.
1. Power on.
1. Run miniterm.

andre-richter, 2021-01-03:
Ah, I didn’t know macOS was more restrictive.
That’s a pity.

Does Miniterm start if you set it to 256000? (You wouldn’t see sane output though, because the Kerne binary is hardcoded to 576000.

andre-richter, 2021-01-03:
I read up a bit on it. Seems macOS usually doesn’t allow more than 230400 when opening the device, but it seems there is a way to circumvent it using stty.

Can you do a test for me?

1. change the baudrate in miniterm.rb back to 230400.
2. Start the Miniterm.
3. In a different console, execute `stty -f /dev/USB_SERIAL_NAME 576000`
4. Only now power the RPi and check if you can see the console output.

sachaos, 2021-01-03:
@andre-richter 

> Does Miniterm start if you set it to 256000? (You wouldn’t see sane output though, because the Kerne binary is hardcoded to 576000.
Yes. It works fine when I revert `ee52e8e288ef6f00416eed94921c2b0e1acfab73`

```
❯ DEV_SERIAL=/dev/tty.usbserial-1420 make miniterm
Miniterm 1.0

[MT] ✅ Serial connected
[0] Booting on: Raspberry Pi 4
[1] Drivers loaded:
      1. BCM GPIO
      2. BCM PL011 UART
[2] Chars written: 93
[3] Echoing input now
kjkdsafdsa

[MT] Bye 👋
```

I'll try the test using stty.

sachaos, 2021-01-03:
@andre-richter It seems `576000` is invalid value on macOS.

```
stty -f /dev/tty.usbserial-1420 576000
stty: tcsetattr: Invalid argument
```

The value `460800` which is mentioned in the stackoverflow was valid.

```
$ stty -f /dev/tty.usbserial-1420 460800

~
$ echo $?
0
```

andre-richter, 2021-01-03:
Thanks!

Then I might need to make the whole tutorials use 460800 to solve this issue and adapt miniterm.rb to spawn the stty when on macOS.

sachaos, 2021-01-03:
@andre-richter I've created PR to fix this problem.
I've checked it works on my macOS.

Please review & merge it If it looks good. #96

andre-richter, 2021-01-04:
Hi @sachaos,

could you please try out https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/pull/97 on your Mac and tell me if it works?
I bumped to 921_600 baud + some more fixes.

I added you as co-author of this new PR to credit your findings.

sachaos, 2021-01-04:
@andre-richter Thank you! I'll try the PR.

commit dfba0f4a (2021-01-04): Bump UART to 921_600 baud + other bugfixes

Fixes #95
Fixes #98
commit c35a30cd (2021-01-04): Bump UART to 921_600 baud + other bugfixes

Fixes #95
Fixes #98
closed 2021-01-04 by andre-richter