rust-embedded/rust-raspberrypi-OS-tutorials #98: Chainload fails sometimes

sachaos, 2021-01-04:
I am trying `07_uart_chainloader`.

I found a problem on `minipush.rb` .
It fails sometimes like below.

```
$ DEV_SERIAL=/dev/tty.usbserial-1420 BSP=rpi4 make chainboot

Minipush 1.0

[MP] ✅ Serial connected
[MP] 🔌 Please power the target now
 __  __ _      _ _                 _
|  \/  (_)_ _ (_) |   ___  __ _ __| |
| |\/| | | ' \| | |__/ _ \/ _` / _` |
|_|  |_|_|_||_|_|____\___/\__,_\__,_|

           Raspberry Pi 4

[ML] Requesting binary

[MP] ⚡ Connection or protocol Error: Remove power and USB serial. Reinsert serial first, then power
```

andre-richter, 2021-01-04:
This should also be fixed by #97 Minipush fixes.

please Review if you like.

sachaos, 2021-01-04:
@andre-richter Great! I've created the PR to fix this bug too.
But I'll close that. Thank you.

andre-richter, 2021-01-04:
What a timing 😂🙈

commit dfba0f4a (2021-01-04): Bump UART to 921_600 baud + other bugfixes

Fixes #95
Fixes #98
commit c35a30cd (2021-01-04): Bump UART to 921_600 baud + other bugfixes

Fixes #95
Fixes #98
closed 2021-01-04 by andre-richter