rust-embedded/rust-raspberrypi-OS-tutorials #101: how about make qemu can also do chainboot test

umaYnit, 2021-01-17:
just use the virtual serial port like tty in linux. i make a simple python scirpt to do that.

```python
#! /usr/bin/env python3
#coding=utf-8

import pty
import os
import select

def mkpty():
    # open tty
    master1, slave = pty.openpty()
    slaveName1 = os.ttyname(slave)
    master2, slave = pty.openpty()
    slaveName2 = os.ttyname(slave)
    print ('\nslave device names: ', slaveName1, slaveName2)
    return master1, master2

if __name__ == "__main__":

    master1, master2 = mkpty()
    while True:
        rl, wl, el = select.select([master1,master2], [], [], 1)
        for master in rl:
            data = os.read(master, 128)
            print ("read %d data." % len(data))
            if master==master1:
                os.write(master2, data)
            else:
                os.write(master1, data)
```

andre-richter, 2021-01-17:
I don't see value in simulating the chainloading, since the chainloader here is explicitly designed to be transparent (you can put each kernel on the SD card directly), and rather a means of convenience for working with the real HW.

It would make sense to have a CI workflow testing the chainloader, though. I'll keep the issue open as a reminder to implement this in the future. Thanks for the suggestion!

By the way, the tests in tutorials 13 and later already make use of pseudo-TTY interfacing to QEMU (in the Ruby test scripts).

umaYnit, 2021-01-17:
thanks for your reply.
I did this because I failed to boot on the real HW, I am looking for why. (I am using  windows and wsl2, forward serial port data through tools, maybe it's a tool problem, so it would be better if miniterm and minipush has a Rust version)

abdes, 2021-04-19:
> ..., so it would be better if miniterm and minipush has a Rust version)

I made one a couple of months ago @umaYnit , if you're interested take a look at https://github.com/setros/bootcom.

chawittt, 2021-04-27:
486931415

andre-richter, 2021-04-29:
@umaYnit you can now do `make test` in the chainloader tutorial and it will test with QEMU.

closed 2021-04-29 by andre-richter

umaYnit, 2021-04-30:
thanks~ :heart:

commit 8982682d (2021-04-29): Add chainloader test

closes #101