rust-embedded/rust-raspberrypi-OS-tutorials #105: Gemfile Error while running bundle install on cloned repo

martinomburajr, 2021-02-22:
Hello! Thanks again for this tutorial, been learning quite a lot.

I cloned the repository and checked into one of the lessons (lesson 6) where we connect to the raspberrypi 4.

I ran the following command as instructed in the `README.md`

```bash
sudo gem install bundler
bundle config set path '.vendor/bundle'
bundle install
```

I continuously get the following error. I have attempted to install ruby dev tools etc, to no avail.

```bash
me@me /me.../rust-raspberrypi-OS-tutorials/06_drivers_gpio_uart > 
bundle install

Fetching gem metadata from https://rubygems.org/.........
Resolving dependencies...
Using ast 2.4.2
Using bundler 2.2.11
Using colorize 0.8.1
Using parallel 1.20.1
Using rainbow 3.0.0
Using regexp_parser 2.0.3
Using rexml 3.2.4
Using ruby-progressbar 1.11.0
Using unicode-display_width 2.0.0
Using parser 3.0.0.0
Using rubocop-ast 1.4.1
Fetching serialport 1.3.1
Using rubocop 1.10.0
Installing serialport 1.3.1 with native extensions
Gem::Ext::BuildError: ERROR: Failed to build gem native extension.

current directory:
/me.../rust-raspberrypi-OS-tutorials/.vendor/bundle/ruby/2.6.0/gems/serialport-1.3.1/ext/native
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby -I /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0 -r
./siteconf20210222-79147-1k74z03.rb extconf.rb
checking for OS... darwin
checking for termios.h... *** extconf.rb failed ***
Could not create Makefile due to some reason, probably lack of necessary
libraries and/or headers.  Check the mkmf.log file for more details.  You may
need configuration options.

Provided configuration options:
        --with-opt-dir
        --without-opt-dir
        --with-opt-include
        --without-opt-include=${opt-dir}/include
        --with-opt-lib
        --without-opt-lib=${opt-dir}/lib
        --with-make-prog
        --without-make-prog
        --srcdir=.
        --curdir
        --ruby=/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/$(RUBY_BASE_NAME)
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:467:in `try_do': The compiler failed to generate an executable file.
(RuntimeError)
You have to install development tools first.
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:585:in `block in try_compile'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:534:in `with_werror'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:585:in `try_compile'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:1109:in `block in have_header'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:959:in `block in checking_for'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:361:in `block (2 levels) in postpone'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:331:in `open'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:361:in `block in postpone'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:331:in `open'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:357:in `postpone'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:958:in `checking_for'
        from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/mkmf.rb:1108:in `have_header'
        from extconf.rb:10:in `<main>'

To see why this extension failed to compile, please check the mkmf.log which can be found here:

/me.../rust-raspberrypi-OS-tutorials/.vendor/bundle/ruby/2.6.0/extensions/universal-darwin-19/2.6.0/serialport-1.3.1/mkmf.log

extconf failed, exit code 1

Gem files will remain installed in
/me.../rust-raspberrypi-OS-tutorials/.vendor/bundle/ruby/2.6.0/gems/serialport-1.3.1 for
inspection.
Results logged to
/me.../rust-raspberrypi-OS-tutorials/.vendor/bundle/ruby/2.6.0/extensions/universal-darwin-19/2.6.0/serialport-1.3.1/gem_make.out

An error occurred while installing serialport (1.3.1), and Bundler cannot continue.
Make sure that `gem install serialport -v '1.3.1' --source 'https://rubygems.org/'` succeeds before bundling.

In Gemfile:
  serialport
me.../rust-raspberrypi-OS-tutorials/06_drivers_gpio_uart > 

```

Here is the output of the `mkmf.log`

```bash
"xcrun clang -o conftest -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/universal-darwin19 -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/backward -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0 -I. -D_XOPEN_SOURCE -D_DARWIN_C_SOURCE -D_DARWIN_UNLIMITED_SELECT -D_REENTRANT    -g -Os -pipe -DHAVE_GCC_ATOMIC_BUILTINS -DOS_DARWIN conftest.c  -L. -L/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib -L. -L/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.Internal.sdk/usr/local/lib   -arch x86_64   -lruby.2.6   "
In file included from conftest.c:1:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby.h:33:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/ruby.h:24:10: fatal error: 'ruby/config.h' file not found
#include "ruby/config.h"
         ^~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/ruby.h:24:10: note: did not find header 'config.h' in framework 'ruby' (loaded from '/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks')
1 error generated.
checked program was:
/* begin */
1: #include "ruby.h"
2: 
3: int main(int argc, char **argv)
4: {
5:   return 0;
6: }
/* end */

```

Not sure if you know what to do with such an error

**OS**: MacOS 10.15.7
**Raspberry PI**: 4

Rust ToolChain
```
stable-x86_64-apple-darwin
nightly-2021-01-08-x86_64-apple-darwin (override)
nightly-x86_64-apple-darwin (default)
```

andre-richter, 2021-02-22:
Hi,

googling this error hints towards broken macOS Ruby installations in Catalina.
Can you check if the "upgrade Ruby to 2.7" instructions in https://github.com/ffi/ffi/issues/653 help?

Best,
Andre

martinomburajr, 2021-02-28:
This seemed to have helped quite a bit. I believe this issue has been resolved.

closed 2021-02-28 by martinomburajr