rust-embedded/rust-raspberrypi-OS-tutorials #109: [Tutorial-6] - Miniterm shows two questions marks after connecting RPI4 to power.

martinomburajr, 2021-03-09:
Hey 👋 and thanks again for this tutorials series.

## Issue Description
I get these questions marks (see images below) after connecting my Raspberry Pi4 to power after completing  [this section](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/06_drivers_gpio_uart#boot-it-from-sd-card) of the tutorial. I've correctly setup the pins. I've loaded the correct files to the SDCard, and have even tried switching between `start.elf` vs `start4.elf`. 

It seems that the Miniterm software does recognize that a device is connected, however it does not shown anything other than two questions marks once I connect the power. I have even cloned and used the tutorial that is in this repo just incase I made a typo or mistake anywhere.

#### Pin Connections
[image]
 
#### Error
```bash
me$ DEV_SERIAL=/dev/tty.usbserial-14220 make miniterm
Miniterm 1.0

me/kernel/.vendor/bundle/ruby/2.7.0/gems/serialport-1.3.1/lib/serialport.rb:25: warning: rb_secure will be removed in Ruby 3.0
[MT] ✅ Serial connected
��

````

#### Screenshot of issue
[image]

andre-richter, 2021-03-11:
From the datasheet, your USB-serial looks good. It supports the `921_600` that is needed.

Maybe there is an issue with the `ruby-serial` used by Miniterm to program the baud rate. A bit of googling turned up this interesting article:
https://www.turais.de/mac-os-x-screen-wont-connect-with-the-correct-baudrate/

Can you please use the binary from `Tutorial 8` (it will be useful because it will constantly output timestamps), and then try to connect to the RPi using the two methods described in the link?

Will be interesting to see if you can see the time stamps appear.

martinomburajr, 2021-03-12:
I attempted running the following. 
1. `cd` into `Tutorial 8`
2. In Terminal 1: I  ran `BSP=rpi4 make && screen /dev/cu.usbserial-14220 921600` in one terminal
3. In Terminal 2: I ran `sudo stty -f /dev/cu.usbserial-14220 921600`
4. I powered the RPI4 and got the following output in Terminal 1.

### Output
```bash
���
```

Still seems to be the same error. I also used the first method outlined in the article and got a `�` response. Does this have anything to do with perhaps the files in the SD card?

andre-richter, 2021-03-13:
After the make, and before launching screen, did you copy kernel8.img on the SD card?

andre-richter, 2021-03-25:
Closing this for now due to no process. I can reopen anytime if needed, just comment here again :+1:

closed 2021-03-25 by andre-richter