rust-embedded/rust-raspberrypi-OS-tutorials #113: question about nightly version

BackMountainDevil, 2021-06-29:
# Env
## rust
- stable-x86_64-unknown-linux-gnu updated - rustc 1.53.0 (53cb7b09b 2021-06-17) (from rustc 1.52.1 (9bc8c42bb 2021-05-09))
- nightly-x86_64-unknown-linux-gnu updated - rustc 1.55.0-nightly (e8cb1a4a5 2021-06-27) (from rustc 1.54.0-nightly (1c6868aa2 2021-05-27))  
- cargo 1.55.0-nightly (9233aa06c 2021-06-22)  
I already have rustup instaled. ` rustup default nightly`  have set the nightly default v
## os
- Linux arch 5.12.13-arch1-2 #1 SMP PREEMPT Fri, 25 Jun 2021 22:56:51 +0000 x86_64 GNU/Linux

# What happened and How?

```bash
$ git clone 
$ cd 01_wait_forever/
$ make doc
\nGenerating docs
info: syncing channel updates for 'nightly-2021-04-25-x86_64-unknown-linux-gnu'
error: no release found for 'nightly-2021-04-25'
make: *** [Makefile:80：doc] 错误 1
[kearney@arch 01_wait_forever]$ cargo install cargo-binutils rustfilt
info: syncing channel updates for 'nightly-2021-04-25-x86_64-unknown-linux-gnu'
error: no release found for 'nightly-2021-04-25'

$ cargo install cargo-binutils rustfilt
info: syncing channel updates for 'nightly-2021-04-25-x86_64-unknown-linux-gnu'
error: no release found for 'nightly-2021-04-25'
```

# solution

I have check the nightly version installed. The I check the [web](https://mirrors.tuna.tsinghua.edu.cn/rustup/dist/) and find out there is no the 'nightly-2021-04-25'. As far as I know, the nightly is very new version beyond stable. So I think the new one should work. Then I see the file named rust-toolchain with git history. It seems that it alway change for the nightly version. So I just change it to the new one.

```bash
- channel = "nightly-2021-04-25"
+ channel = "nightly-2021-06-27"
```

```
$ cargo --version
info: syncing channel updates for 'nightly-2021-06-27-x86_64-unknown-linux-gnu'
warning: Signature verification failed for 'https://mirrors.tuna.tsinghua.edu.cn/rustup/dist/2021-06-27/channel-rust-nightly.toml'
info: latest update on 2021-06-27, rust version 1.55.0-nightly (a1411de9d 2021-06-26)
info: downloading component 'cargo'
info: downloading component 'clippy'
info: downloading component 'llvm-tools-preview'
info: downloading component 'rust-docs'
info: downloading component 'rust-std' for 'aarch64-unknown-none-softfloat'
info: downloading component 'rust-std'
info: downloading component 'rustc'
info: downloading component 'rustfmt'
info: installing component 'cargo'
info: installing component 'clippy'
info: installing component 'llvm-tools-preview'
info: installing component 'rust-docs'
info: installing component 'rust-std' for 'aarch64-unknown-none-softfloat'
info: installing component 'rust-std'
info: installing component 'rustc'
info: installing component 'rustfmt'
cargo 1.55.0-nightly (9233aa06c 2021-06-22)

$ cargo install cargo-binutils rustfilt

$ make doc
\nGenerating docs
   Compiling mingo v0.1.0 (/home/kearney/Documents/code/rust/rust-raspberrypi-OS-tutorials/01_wait_forever)
 Documenting mingo v0.1.0 (/home/kearney/Documents/code/rust/rust-raspberrypi-OS-tutorials/01_wait_forever)
    Finished release [optimized] target(s) in 1.22s
     Opening /home/kearney/Documents/code/rust/rust-raspberrypi-OS-tutorials/01_wait_forever/target/aarch64-unknown-none-softfloat/doc/kernel/index.html
```

# expect

can the nightly version just let it be null? it should be smart to know its newest version.

andre-richter, 2021-06-30:
Hi,

we are currently pinning the nightly to a specific version in the `rust-toolchain` file. This is done for (i) keeping it consistent between tutorials and users, and (ii) recently also because newer nightlies are incompatible with some dependencies we have in later tutorials (tock-registers doesn't compile on newer nightlies yet. A new release is due soon, though.

The Github Actions of this tutorial check every night if downloading the pinned nightly works, and it worked for this night too: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/actions

I plan to stay at the pinned nightly until the new tock-registers release is out and I had time to switch to it.

BackMountainDevil, 2021-06-30:
It is very nice for your reply. Now I know why. Thank you all the time.

closed 2021-06-30 by BackMountainDevil

BackMountainDevil, 2021-06-30:
By the way, cause I use [TUNA](https://mirror.tuna.tsinghua.edu.cn/help/rustup/) as a rusup mirror. And they just keep nihtly for a while. This is the reason why I can't down the version of nightly-2021-04-25-x86_64-unknown-linux-gnu. So just cancel the TUNA mirror for now  before download the nightly-2021-04-25.