rust-embedded/rust-raspberrypi-OS-tutorials #127: Makefile Aarch64 Issues

WittsEnd2, 2021-11-04:
When trying to run `make qemu`, it seems to error out on line 129 (at least in part 03_hacky_hello_world)
```
    @$(DOCKER_QEMU) $(EXEC_QEMU) $(QEMU_RELEASE_ARGS) -kernel $(KERNEL_BIN) 
```
Is there a way to fix this so that it works with Aarch64 (especially useful for those that have M1 chips). 

I am working on figuring this out myself too, but would appreciate any help I can get.

andre-richter, 2021-11-04:
Hmm yeah, currently everything assumes an x86 host machine. The docker image is definitely x86 only at the moment.
I don’t know if docker on M1 is already supported.

This might be something I can try to address during this year’s holiday season. 

For now, you can try to modify the Makefile so that it doesn’t try to launch stuff through docker, and then get a native version of `qemu-system-aarch64` installed on your Mac.

Not sure what the best way to do this is. Manual compilation should always work. Maybe there’s a shortcut through Homebrew as well?

cc-morning, 2021-11-20:
My suggestion is to create an archlinux arm on m1 via qemu to run this project, just pass the serial device to qemu and replace the x86_64 gcc in docker with aarch64 gcc and he will work perfectly, that's what I'm doing now.

Example:

[image]

andre-richter, 2021-11-20:
I think what I will be aiming for, once I get around to doing it, is to provide an Ubuntu-based ARM64 container for everything except Make targets that require the USB serial, like chainboot.

These should run natively on Macs with little effort, since Ruby is preinstalled on macOS anyways. So it should just be a matter of installing the required gems.

WittsEnd2, 2021-11-20:
I also learned that there is something in Docker called BuildX which allows for building cross-compiled containers. Not sure how it works though

andre-richter, 2021-11-22:
Fixed by https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/commit/89472abea5244f691874a158ad4c1ca95483b0cc

closed 2021-11-24 by andre-richter