rust-embedded/rust-raspberrypi-OS-tutorials #137: Does this `transition_to_single_core_main` work on the rpi4?

nihalpasham, 2022-01-08:
I guess this is more of a question than an issue but I've found that the rpi4 doesnt produce the expected result when using atomic operations. 

So, I have 2 atomic operations lined up one after the other. I'd have expected an Ok(1) after the compare and swap.
``` rust
    info!("kernel_state: {:?}", state::state_manager().0.load(Ordering::Acquire));
    info!("compare and swap {:?}", state::state_manager().0
            .compare_exchange(
                state::StateManager::INIT,
                state::StateManager::SINGLE_CORE_MAIN,
                Ordering::Acquire,
                Ordering::Relaxed,
            ));
```
but instead I get an error.

```sh
[    0.481485] kernel_state: 0
[    0.484178] compare and swap Err(1)
```
A note: The MMU is disabled. I'm not familiar with atomic instructions but for atomic operations to work on the rpi4, must the MMU  be enabled. Is this correct?  I was going to test this by enabling the MMU but just wanted to check before I head down this road.

andre-richter, 2022-01-08:
MMU must be enabled.

See: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/discussions/132

nihalpasham, 2022-01-08:
Thank you. I must say - (for an opensource tutorial) the turn around time on a response here is awesome (something I wish more projects emulated).

Thanks again 👍

closed 2022-01-08 by nihalpasham