rust-embedded/rust-raspberrypi-OS-tutorials #141: a stupid question: how to boot up the qemu.

lllsssskkk, 2022-01-26:
i've been encountered a problem every time i run the `make qemu` . 

[image]

i really  dont think such image exists in the docker hub.

any hints guide me through?pls

andre-richter, 2022-01-27:
What is your execution environment?
the `\n` that get printed look like there might be something odd.

Can you show what gets printed when you remove the `@` in `@$(DOCKER_QEMU) $(EXEC_QEMU) $(QEMU_RELEASE_ARGS) -kernel $(KERNEL_BIN)` in `Makefile`.

lllsssskkk, 2022-01-28:
Hey 
   thx for your reply. The followings are bunches of images, hope they are relevant to the question.

my system info :
[image]

my docker version: 
[image]

my rust version 
[image]

after deleting the at symbol. make qemu code in the makefile :
[image]

the output of running `make qemu` with modified makefile

[image]

just another silly question.

In an ideal situation, all we'd have to do to start the qemu is install docker and rust and run'make qemu' in each folder. 

Is there anything else I should do before calling'make qemu'?

andre-richter, 2022-01-28:
The emitted command line looks correct. Maybe your local docker setup is in a bad state.

Can you try ‚`docker logout` once before testing again.

lllsssskkk, 2022-01-29:
unfortunately, the problem remains after logging out. The output is still the same, complaining cant pull the year image

andre-richter, 2022-01-30:
What happens when you do `docker pull rustembedded/osdev-utils:2021.12` in your terminal?

lllsssskkk, 2022-01-30:
I can pull the image without errors

[image]

Is there a way in Docker to build the container using an existing image rather than pulling it from the hub again?

andre-richter, 2022-01-30:
There aren’t many options left where it goes wrong.

the print of the docker command that make generates looks good. Could can pull the image when invoking docker manually.

The printing of `\n` indicates something with the shell in which make gets executed.
Which shell are you using?

can you execute `bash` once before executing `make`. Does that help?

lllsssskkk, 2022-02-01:
If it looks like to me that only my machine has this kinda of problem, i must set something wrong in my env. 

As i realised i have all tools needed to build the image. I decide to manually load the image into the qemu.  Since i figure out how to load the image with qemu, i decide to close this issue. Thx for your help, :D.

closed 2022-02-01 by lllsssskkk