rust-embedded/rust-raspberrypi-OS-tutorials #163: invalid file operation error

EWYan, 2022-05-10:
i just try to compile 01_wait_forever samples on ubuntu1604, but don't know why the following error pops out:
```
yan@ubuntu:~/Desktop/rust-raspberrypi-OS-tutorials/01_wait_forever$ make qemu
Makefile:120: *** Invalid file operation: < target/aarch64-unknown-none-softfloat/release/kernel.d.  Stop.
```
please help, thx.

andre-richter, 2022-05-10:
Hmm interesting. Did you change anything or is every file untouched?

What is the output of `make —version`?

andre-richter, 2022-05-10:
Probably a make version that is too old. Can you try this as a workaround for now: 

https://stackoverflow.com/questions/6767413/create-a-variable-in-a-makefile-by-reading-contents-of-another-file#6779306

EWYan, 2022-05-11:
> Probably a make version that is too old. Can you try this as a workaround for now:
> 
> https://stackoverflow.com/questions/6767413/create-a-variable-in-a-makefile-by-reading-contents-of-another-file#6779306

yes, make 4.1 is not works for this file operation, make 4.3 works like charm; thanks.

closed 2022-05-11 by EWYan