rust-embedded/rust-raspberrypi-OS-tutorials #164: error[E0463]: can't find crate for `core`

EWYan, 2022-05-11:
hi, there,
i tried to compile 01_wait_forever sample but get an error log like below:
```
yan@ubuntu:~/Desktop/rust-raspberrypi-OS-tutorials/01_wait_forever$ sudo make qemu

Compiling kernel ELF - rpi3
   Compiling mingo v0.1.0 (/home/yan/Desktop/rust-raspberrypi-OS-tutorials/01_wait_forever)
error[E0463]: can't find crate for `core`
  |
  = note: the `aarch64-unknown-none-softfloat` target may not be installed
  = help: consider downloading the target with `rustup target add aarch64-unknown-none-softfloat`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `mingo` due to previous error
make: *** [Makefile:121: target/aarch64-unknown-none-softfloat/release/kernel] Error 101

```

according to log note "consider downloading the target with `rustup target add aarch64-unknown-none-softfloat`",
but i think the mentioned compiler had been added on my environment:
```
yan@ubuntu:~/Desktop/rust-raspberrypi-OS-tutorials/01_wait_forever$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/yan/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
nightly-2022-04-10-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu (default)

installed targets for active toolchain
--------------------------------------

aarch64-unknown-none
aarch64-unknown-none-softfloat
x86_64-unknown-linux-gnu

active toolchain
----------------

nightly-2022-04-10-x86_64-unknown-linux-gnu (overridden by '/home/yan/Desktop/rust-raspberrypi-OS-tutorials/rust-toolchain.toml')
rustc 1.62.0-nightly (8bf93e9b6 2022-04-09)
```
so, may something in my  environment  setting was not right, but don't know how to figure this mess out.
please give some advice, thx.

andre-richter, 2022-05-11:
Should be the `sudo`. It makes it use root installation of rust which will not have the target.

EWYan, 2022-05-11:
thx for reply.
is the follow print log as expected?
it seems not like the description in project MD files
```
yan@ubuntu:~/Desktop/rust-raspberrypi-OS-tutorials/01_wait_forever$ make qemu

Launching QEMU
----------------
IN: 
0x00000000:  580000c0  ldr      x0, #0x18
0x00000004:  aa1f03e1  mov      x1, xzr
0x00000008:  aa1f03e2  mov      x2, xzr
0x0000000c:  aa1f03e3  mov      x3, xzr
0x00000010:  58000084  ldr      x4, #0x20
0x00000014:  d61f0080  br       x4

----------------
IN: 
0x00000300:  d2801b05  mov      x5, #0xd8
0x00000304:  d53800a6  mrs      x6, mpidr_el1
0x00000308:  924004c6  and      x6, x6, #3
0x0000030c:  d503205f  wfe      
0x00000310:  f86678a4  ldr      x4, [x5, x6, lsl #3]
0x00000314:  b4ffffc4  cbz      x4, #0x30c

----------------
IN: 
0x0000030c:  d503205f  wfe      
0x00000310:  f86678a4  ldr      x4, [x5, x6, lsl #3]
0x00000314:  b4ffffc4  cbz      x4, #0x30c

----------------
IN: 
0x00000300:  d2801b05  mov      x5, #0xd8
0x00000304:  d53800a6  mrs      x6, mpidr_el1
0x00000308:  924004c6  and      x6, x6, #3
0x0000030c:  d503205f  wfe      
0x00000310:  f86678a4  ldr      x4, [x5, x6, lsl #3]
0x00000314:  b4ffffc4  cbz      x4, #0x30c

----------------
IN: 
0x00080000:  d503205f  wfe      
0x00080004:  17ffffff  b        #0x80000

----------------
IN: 
0x00000300:  d2801b05  mov      x5, #0xd8
0x00000304:  d53800a6  mrs      x6, mpidr_el1
0x00000308:  924004c6  and      x6, x6, #3
0x0000030c:  d503205f  wfe      
0x00000310:  f86678a4  ldr      x4, [x5, x6, lsl #3]
0x00000314:  b4ffffc4  cbz      x4, #0x30c
```