rust-embedded/rust-raspberrypi-OS-tutorials #166: Complete Chinese Translation

Jzow, 2022-08-09:
This issue contains the description of the entire Chinese translation and the description of the pull request.

The following is the list to be translated. If there is any omission, please supplement

- [ ] [03_hacky_hello_world](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/03_hacky_hello_world)
- [ ] [04_safe_globals](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/04_safe_globals)
- [ ] [05_drivers_gpio_uart](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/05_drivers_gpio_uart)
- [ ] [06_uart_chainloader](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/06_uart_chainloader)
- [ ] [07_timestamps](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/07_timestamps)
- [ ] [08_hw_debug_JTAG](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/08_hw_debug_JTAG)
- [ ] [09_privilege_level](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/09_privilege_level)
- [ ] [10_virtual_mem_part1_identity_mapping](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/10_virtual_mem_part1_identity_mapping)
- [ ] [11_exceptions_part1_groundwork](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/11_exceptions_part1_groundwork)
- [ ] [12_integrated_testing](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/12_integrated_testing)
- [ ] [13_exceptions_part2_peripheral_IRQs](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/13_exceptions_part2_peripheral_IRQs)
- [ ] [14_virtual_mem_part2_mmio_remap](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/14_virtual_mem_part2_mmio_remap)
- [ ] [15_virtual_mem_part3_precomputed_tables](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/15_virtual_mem_part3_precomputed_tables)
- [ ] [16_virtual_mem_part4_higher_half_kernel](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/16_virtual_mem_part4_higher_half_kernel)
- [ ] [17_kernel_symbols](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/17_kernel_symbols)
- [ ] [18_backtrace](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/18_backtrace)
- [ ] [19_kernel_heap](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/tree/master/19_kernel_heap)

Phantato, 2023-02-16:
Hi, I'm willing to help with the translation. I've noticed that there is an unmerged PR for chpt.4 so I thought I could begin with chpt.5?

Jzow, 2023-02-17:
@posborne sure, It is necessary to open new PR corresponding to different chapters

eldruin, 2023-02-17:
@Phantato You could also review the chapter 4 PR #171 and viceversa and so you can help each other.

Jzow, 2025-03-13:
Sorry, I forgot about this task. I will spend some time translating it next.