rust-embedded/rust-raspberrypi-OS-tutorials #178: everything seems to work fine but I cant input anything

thivmcthiv, 2023-05-23:
My os: Ubuntu 22

Problem:
I cant input anything, and when I type in characters to the console, nothing is echoed.

I tried running the kernel in qemu, and on my rpi4, but neither worked.

Example output that seems to be working fine from 09:

```
[    0.208975] mingo version 0.9.0
[    0.209061] Booting on: Raspberry Pi 4
[    0.209462] Current privilege level: EL1
[    0.209938] Exception handling state:
[    0.210382]       Debug:  Masked
[    0.210772]       SError: Masked
[    0.211162]       IRQ:    Masked
[    0.211552]       FIQ:    Masked
[    0.211942] Architectural timer resolution: 18 ns
[    0.212516] Drivers loaded:
[    0.212853]       1. BCM PL011 UART
[    0.213275]       2. BCM GPIO
[    0.213633] Timer test, spinning for 1 second
[    1.214163] Echoing input now

```

I thought maybe it was just my terminal so I wrote some rust to try sending bytes to /dev/ttyUSB0 where the rpi4 was connected:

```
//using serialport 3.0.0
fn write_to_serial_device(input: &str) {
    let port_name = "/dev/ttyUSB0";
    let mut settings: SerialPortSettings = Default::default();
    settings.timeout = Duration::from_millis(100);
    settings.baud_rate = 9600;
    match serialport::open_with_settings(port_name, &settings) {
        Ok(mut port) => {
            if let Err(e) = writeln!(port, "{}", input) {
                eprintln!("Error writing to serial port {}: {}", port_name, e);
            } else {
                println!("Successfully wrote to serial port {}", port_name);
            }
        }
        Err(e) => {
            eprintln!("Error opening serial port {}: {}", port_name, e);
        }
    }
}
```
that code didnt report any errors when executing, but still nothing was echoed in the console after running it.

fair warning, It is entirely possible that I am just misunderstanding something or making a dumb mistake because I have never worked with embedded rust before this so let me know if I shouldn't have opened this issue :))

Tim-C1, 2023-05-27:
I think you should test your uart on your rpi4 and your usb-to-tty converter works well, I encounter the same problem when I purchase a old rpi3.

yonchicy, 2023-12-09:
do you solve it ? I got the same problem on pi 3