rust-embedded/rust-raspberrypi-OS-tutorials #184: Errors regarding Github CI/CD

Jzow, 2023-09-09:
I have noticed that every time a PR is performed, the Github CI/CD program performs status checks, and it seems to report an issue with copyright years in the Makefile file

example:

```shell
Clippy X1_JTAG_boot - BSP: rpi4
   Compiling mingo v0.8.0 (/home/runner/work/rust-raspberrypi-OS-tutorials/rust-raspberrypi-OS-tutorials/X1_JTAG_boot)
    Checking tock-registers v0.8.1
    Checking aarch64-cpu v9.0.0
    Finished release [optimized] target(s) in 2.17s
Checking for copyright: 01_wait_forever/Makefile
	Only found years: [2018, 2022]
Error: Process completed with exit code 1.
```

https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/49a25275e66ab0dcacfcbbbb666311df7a9650c9/01_wait_forever/Makefile#L3

But I'm not sure if it caused it error.

Jzow, 2023-09-09:
#188

eldruin, 2023-09-12:
It is probably complaining that `2023` is not present. Adding it may fix it.

Jzow, 2023-09-12:
> It is probably complaining that `2023` is not present. Adding it may fix it.

Okay, I'll add it later and resubmit the PR.

closed 2023-09-15 by Jzow