rust-embedded/rust-raspberrypi-OS-tutorials #197: Depracated link in The tl;dr Version subsection of README.md file

nmpluta, 2023-09-26:
The link provided in the [🚀 Version tl;dr](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/README.md#-the-tldr-version) section `System Requirements` in `README.md` seems to be deprecated. It concerns the link to `rust-toolchain` mentioned in point 3: 
> Prepare the Rust toolchain. Most of it will be handled on first use through the [rust-toolchain](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/rust-toolchain) file. What's left for us to do is:

[image]

eldruin, 2023-09-27:
Nice find. The `.toml` part of the URL got removed. 
The right URL is this one: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/rust-toolchain.toml
Could you submit a PR fixing this?

nmpluta, 2023-09-30:
> Nice find. The `.toml` part of the URL got removed. The right URL is this one: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/rust-toolchain.toml Could you submit a PR fixing this?

Sure. Here's PR: #199

PR #199 (2023-10-02): fix depracated link to the rust-toolchain.toml file
### Description

The link provided in the [🚀 Version tl;dr](https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/master/README.md#-the-tldr-version) section System Requirements in README.md seems was deprecated. The issue concerns also the `README.ES.md`.

Related Issue:  https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/issues/197
     
### Pre-commit steps - response:
This is just a README change, only `./devtool ready_for_publish_no_rust` was run.

Fixes #197
closed 2023-10-02 by eldruin