rust-embedded/rust-raspberrypi-OS-tutorials #200: Invalid links in 11_exceptions_part1_groundwork/README.md

Tartasprint, 2023-10-03:
In the README.MD of chapter 11 there are some references to the "Arm Architecture Reference Manual for A-profile architecture":

https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/53074bb82244d912fa85af51d92b17d829692fad/11_exceptions_part1_groundwork/README.md?plain=1#L58

https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/53074bb82244d912fa85af51d92b17d829692fad/11_exceptions_part1_groundwork/README.md?plain=1#L68-L69

with the `[ARMv8_Manual]` referring to:

https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/53074bb82244d912fa85af51d92b17d829692fad/11_exceptions_part1_groundwork/README.md?plain=1#L87

The problem is that this link refers to the latest version of the manual, but the text is referring to another version. I went through various versions of the ARM manual but couldn't find the one being referred to.

In older versions (such as E.a) section D1.10.1 is "Preferred exception return address" which should be similar to page 5357 in the [current latest version of the document](https://developer.arm.com/documentation/ddi0487/ja).

I couldn't find what the mentioned pages 1874-1876 were referring to.

I suggest picking a version of the manual and write the references to it with a more permanent link than using "latest" version, to avoid such problems.

While I'm there, I thank Andre Richter and everyone else who has contributed to these fantastic tutorials ! :partying_face:

Jzow, 2023-10-04:
Hi @Tartasprint I checked question about [a-profile_architecture_reference_manual.pdf](https://developer.arm.com/documentation/ddi0487) you mentioned.

My simple understanding of the question you raised is that we need to specify the page number in the document corresponding to a specific version of the manual to solve the problem of not finding the corresponding page when updating the new version of the ARM PDF manual.

Tartasprint, 2023-10-12:
Hi, yes that is what I meant, thanks for the fix !

closed 2023-10-12 by Tartasprint