rust-embedded/rust-raspberrypi-OS-tutorials #205: Plans for even more embedded devices

FairyTail2000, 2023-11-13:
Are there any plans for supporting devices like the Pico WH, Pico W?

These are decently smaller than the rpi 4 but share a similar cpu and gpu layout. They also don't have an sdcard so flashing is done a little bit differently.

Do you have any idea if and when you could do someting like that?

ChocolateLoverRaj, 2026-02-21:
I think `embassy-rp` covers most use cases for Raspberry Pi Picos. See https://github.com/embassy-rs/embassy/tree/main/examples/rp/src/bin.

jonathanpallant, 2026-02-22:
> These are decently smaller than the rpi 4 but share a similar cpu and gpu layout.

The Pico devices are Cortex-M microcontrollers that have basically nothing in common with the larger Raspberry Pi boards.

(OK, except that a Raspberry Pi 5's I/O southbridge - the RP1 - is technically a bit similar to the RP2 on a Raspberry Pi Pico, but there is no SDK for programming the RP1 independently so it's moot).

You perhaps mean the Raspberry Pi Zero, which has broadly the same SoC as a Raspberry Pi 1B+, or the Raspberry Pi Zero 2, which has broadly the same SoC as the Raspberry Pi 3.