rust-embedded/rust-raspberrypi-OS-tutorials #214: Problem setting up the MacOS environment, the serialport gem is not maintained.

livexia, 2024-05-02:
Problem: Unable to use bundle to install Ruby gem [ruby-serialport](https://github.com/hparra/ruby-serialport)

System info: Apple M1 with macOS Sonoma

ruby-serialport issue: [posix_serialport_impl.c: error: incompatible pointer to integer](posix_serialport_impl.c: error: incompatible pointer to integer)

The serialport gem is no longer maintained. I can pull the repo and apply the patch mentioned in the issue, but due to my lack of knowledge of Ruby/gem, it's not perfect, for some reason  I need to add `RUBYLIB=ruby-serialport/lib` to specify the serialport lib path. Maybe someone good at gem has a better solution.I wonder if this can be done with rust.

listentodella, 2024-06-02:
Any better solutions ? I also met this issue.

livexia, 2024-06-02:
A better solution would be to use a maintained serialport gem to replace the current one. If you just want to run the tutorials, I think manual patch ruby-serialport and install will do, but if you are not familiar with ruby, you will need to spend some time figuring out how to install the gem locally.

links: 

- https://github.com/hparra/ruby-serialport/issues/74

listentodella, 2024-06-02:
yes, it's hard to know how to install gem locally.

livexia, 2024-06-02:
0. `bundle config --local path '.vendor/bundle'`
1. run `gem install --local pkg/serialport-1.3.2.gem --install-dir ../ruby-serialport-output -V` intall gem to `ruby-serialport-output`
2. in Gemfile using `gem 'serialport', path: "ruby-serialport-output"` to specify path
3. bundle install 
4. bundle list
```
❯ bundle list
Gems included by the bundle:
  * bindata (2.5.0)
  * bundler (1.17.2)
  * colorize (1.1.0)
  * elftools (1.2.0)
  * ruby-progressbar (1.13.0)
  * serialport (1.3.2)
 ```

BartMassey, 2024-06-02:
Somebody give us a pull request for the book? This is something I know nothing about, but it sounds like @livexia has it figured out.

livexia, 2024-06-02:
I don't think this solution is good or 100% correct, even though it works on my system. I came across this problem because I am helping someone run this tutorial. It fails on my own system, but had no problem on that person's system. Before this I know nothing about ruby, I think this is a good temporary solution, maybe someone with better knowledge should give a pr.

Jzow, 2024-07-30:
Thank you for reporting this issue. The author has been on vacation for over a year now.

This seems to be a Ruby issue, I will take some time to check, but I'm not sure if I can solve it, but it's worth fixing it.