rust-embedded/rust-raspberrypi-OS-tutorials #217: Unable to set correct baud rate for Uart (Raspi 3)

FallenStarAngelFox, 2024-08-20:
I am having issues with setting up uart baudrate on raspberry pi 3, I have attempted to set different baud rates but it keeps printing `Set PL011 baud rate to 103448.300000 Hz`  i am using  `minicom -b 115200 -D /dev/tty.usbserial-0001` with 2nd_stage_uart=1,enable_uart=1 to debug
```
Raspberry Pi Bootcode
Read File: config.txt, 66
Read File: start.elf, 2981376 (bytes)
Read File: fixup.dat, 7303 (bytes)
MESS:00:00:01.056295:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.060416:0: brfs: File read: 66 bytes
MESS:00:00:01.097094:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.103254:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.109591:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.115929:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.122265:0: HDMI0:EDID error reading EDID block 0 attempt 4         
MESS:00:00:01.128602:0: HDMI0:EDID error reading EDID block 0 attempt 5         
MESS:00:00:01.134938:0: HDMI0:EDID error reading EDID block 0 attempt 6         
MESS:00:00:01.141275:0: HDMI0:EDID error reading EDID block 0 attempt 7         
MESS:00:00:01.147612:0: HDMI0:EDID error reading EDID block 0 attempt 8         
MESS:00:00:01.153949:0: HDMI0:EDID error reading EDID block 0 attempt 9         
MESS:00:00:01.160044:0: HDMI0:EDID giving up on reading EDID block 0            
MESS:00:00:01.166799:0: brfs: File read: /mfs/sd/config.txt                     
MESS:00:00:01.170993:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defind
MESS:00:00:01.371464:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not d
MESS:00:00:01.378667:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defind
MESS:00:00:01.384928:0: *** Restart logging                                     
MESS:00:00:01.388801:0: brfs: File read: 66 bytes                               
MESS:00:00:01.394371:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0   
MESS:00:00:01.401330:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1   
MESS:00:00:01.408189:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2   
MESS:00:00:01.415047:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3   
MESS:00:00:01.421904:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4   
MESS:00:00:01.428762:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5   
MESS:00:00:01.435620:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6   
MESS:00:00:01.442477:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7   
MESS:00:00:01.449335:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8   
MESS:00:00:01.456193:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9   
MESS:00:00:01.462807:0: hdmi: HDMI0:EDID giving up on reading EDID block 0      
MESS:00:00:01.468712:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0   
MESS:00:00:01.476506:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1   
MESS:00:00:01.483363:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2   
MESS:00:00:01.490220:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3   
MESS:00:00:01.497078:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4   
MESS:00:00:01.503936:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5   
MESS:00:00:01.510794:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6   
MESS:00:00:01.517651:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7   
MESS:00:00:01.524509:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8   
MESS:00:00:01.531366:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9   
MESS:00:00:01.537981:0: hdmi: HDMI0:EDID giving up on reading EDID block 0      
MESS:00:00:01.543853:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_dd
MESS:00:00:01.552342:0: HDMI0: hdmi_pixel_encoding: 162000000                   
MESS:00:00:01.558056:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-0
MESS:00:00:01.585450:0: dtb_file 'bcm2710-rpi-3-b.dtb'                          
MESS:00:00:01.589452:0: dtb_file 'bcm2837-rpi-3-b.dtb'                          
MESS:00:00:01.594312:0: dterror: Failed to load Device Tree file '?'            
MESS:00:00:01.600389:0: Failed to open command line file 'cmdline.txt'          
MESS:00:00:01.678441:0: brfs: File read: /mfs/sd/kernel8.img                    
MESS:00:00:01.682407:0: Loaded 'kernel8.img' to 0x200000 size 0xee4c0           
MESS:00:00:01.689877:0: Kernel relocated to 0x80000                             
MESS:00:00:01.693229:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defind
MESS:00:00:01.702244:0: uart: Set PL011 baud rate to 103448.300000 Hz           
MESS:00:00:01.708536:0: uart: Baud rate change done...                          
MESS:00:00:01.711967:0: uart: Baud rate
```

BartMassey, 2024-08-20:
Possible the Pi 3 UART won't do 115200 and thus sets the baud rate to "as fast as it can go"? Try setting things up for 57600 and see if it all works.

FallenStarAngelFox, 2024-08-20:
Tried that as well and ended up getting the same results, I'm going to compile just the uart_init and test that in case something is causing a panic before reaching that point

FallenStarAngelFox, 2024-08-25:
Somehow fixed it by going through and revising all the code, I think it was the way the kernel was compiled through linker arguments that caused the problem

closed 2024-08-25 by FallenStarAngelFox

BartMassey, 2024-08-25:
Glad you got it working. Anything that you can share about what made it work would be helpful.

FallenStarAngelFox, 2024-08-25:
I am not sure because I reworked all or my code, I think it was mainly because I only compiled using cargo build, and because I wasn't using rust-objcopy to output a binary